vmatare/thinkfan #96: Merging multifan branch breaks config file

jthomme1, 2020-06-09:
Hi, I'm using thinkfan on my t490 and the last commit (aka the merging of the multifan branch 6f294483da5c426f1bee0eb7f46302113aca51e9) broke the option
```
- fans:
  - tpacpi: /proc/acpi/ibm/fan
```
When trying to start thinkfan, I get the error message: 
```
Jun 09 18:08:30 t490 thinkfan[761073]: ERROR: /etc/thinkfan.conf:47:
                                         - tpacpi: /proc/acpi/ibm/fan
                                         ^
                                       Invalid entry.
```
Reverting to the commit 80056f9a9e9474b66001aa139d42c4d1a3e411ff I do not have this problem, so it definitively comes from this merge.
Hope this helps finding the issue.
Greetings

vmatare, 2020-06-09:
I can't reproduce this without the full config. Can you post it here please?

jthomme1, 2020-06-12:
Here is my exact config file:
```
sensors:

  - hwmon: /sys/devices/platform/thinkpad_hwmon/hwmon/
    name: thinkpad
    # skipping index 11 because it sometimes shows this weird 66° without reason
    # skipping index 2 because it disappears, 8 also makes problems
    # just leaving index 1 and 5 because they are the only ones with values
    # and 9 - 16 have the weird habit to only sometimes show up
    indices: [1, 5]
    optional: true

  - hwmon: /sys/devices/platform/coretemp.0/hwmon/
    name: coretemp
    indices: [1, 2, 3, 4, 5]
    optional: true

  - hwmon: /sys/devices/pci0000:00/0000:00:1d.4/0000:3d:00.0/hwmon/
    name: nvme
    indices: [1, 2, 3]
    optional: true

  - hwmon: /sys/devices/virtual/thermal/thermal_zone0/hwmon0/
    name: acpitz
    indices: [1]
    optional: true

  - hwmon: /sys/devices/virtual/thermal/
    name: iwlwifi_1
    # skipping index 11 because it sometimes shows this weird 66° without reason
    indices: [1]
    optional: true

  - hwmon: /sys/devices/virtual/thermal/
    name: pch_cannonlake
    indices: [1]
    optional: true

#  - hwmon: /sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:19/PNP0C09:00/PNP0C0A:00/power_supply/BAT0/
#    name: BAT0
#    indices: [1]
#
#  - hwmon: /sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:19/PNP0C09:00/ACPI0003:00/power_supply/AC/
#    name: AC
#    indices: [1]

fans:
  - tpacpi: /proc/acpi/ibm/fan

# Simple level config
# meaning: [FANSPEED, LOWER_LIMIT, UPPER_LIMIT]
#
levels:
   - [0, 0, 35]
   - [1, 35, 40]
   - [2, 40, 45]
   - [3, 45, 50]
   - [4, 50, 55]
   - [5, 55, 60]
   - [6, 60, 65]
   - [7, 65, 32767]

```
I just double checked, I get the error only with the latest commit (merge).

gogron, 2020-07-04:
similar failure on thinkpad x220 after updating to 1.2

`thinkfan[2225]: ERROR: /proc/acpi/ibm/fan: Unknown file format.`

```
/proc/acpi/ibm/fan
status:		enabled
speed:		2936
level:		auto
commands:	level <level> (<level> is 0-7, auto, disengaged, full-speed)
commands:	enable, disable
commands:	watchdog <timeout> (<timeout> is 0 (off), 1-120 (seconds))
```
```
/etc/thinkfan.yaml
sensors:
  - hwmon: /sys/class/thermal/thermal_zone0/temp
  - tpacpi: /proc/acpi/ibm/fan
levels:
  - [0, 0, 50]
  - ["level auto", 45, 75]
  - ["level disengaged", 70, 255]
```

vmatare, 2020-07-12:
Sorry this took a while, but the support for multiple fans brings some significant changes to the config structure, which makes it more difficult to properly cover all the use cases. Anyways, fix is underway.

vmatare, 2020-07-12:
tracking this and a bunch of other fixes in PR #102. Pending some more tests on my side, but you can try it out if you want to help.

vmatare, 2020-07-12:
Merged into master, please reopen if you find it's not fixed. Thanks for reporting.

closed 2020-07-12 by vmatare

jthomme1, 2020-07-13:
It indeed appears to be fixed, thanks for working on it!