vmatare/thinkfan #97: windows build?

4u7157, 2020-06-26:
Hello would it be possible to port this to windows?
Thank you

vmatare, 2020-06-26:
Well. I was about to give you a plain "no". But while I'm not inclined to look into it, somebody else might be, so here's the long answer:

First of all, I'm not familiar at all with the infrastructure Windows offers for this kind of thing, so I'm not sure about any of this.

However _as far as I know_, the problem on Windows is that there is no unified interface for accessing temperature sensors and/or to control fans. That is why you end up with these crappy hardware monitoring tools made by each manufacturer independently, with the Asus tool controlling only the Asus fans and MSI tool controlling only the MSI fans.

Now thinkfan uses the Linux **sysfs** to basically do everything. In sysfs, every temperature sensor and every controllable fan is simply a file you can read from and/or write to. This is the unified interface I'm talking about and it's the reason we don't have to write any hardware-specific code. Now in theory it is of course possible to give thinkfan a bazillion different hardware drivers for every GPU, mainboard or whatever that's out there. But it's precisely this lack of standardization and coherence that is the reason I'm not using Windows. But on the other hand, there's of course a lot more manpower there, so doing all the hardware interfacing manually might actually be feasible if enough people come together.

4u7157, 2020-06-27:
okay thank you for your answer

yuceltoluyag, 2020-07-05:
https://sourceforge.net/projects/tp4xfancontrol/ @4u7157  you can use this

vmatare, 2020-07-31:
Just stumbled across this: https://github.com/hirschmann/nbfc
It appears they are actually going through the hardware-specific PITA I mentioned, and they even manage to do it portably on both Windows and Linux. It's probably not as awesome as thinkfan ;-), but it might be interesting for Windows users and for people lacking Linux hwmon driver support.

closed 2020-07-31 by pr0f35510n4l

antoon-r, 2021-12-15:
I can confirm that https://github.com/hirschmann/nbfc worked for me - at least I was able to speed my fan to 100% in Windows, Lenovo T15 (used Lenovo ThinkPad T430s profile)