vmatare/thinkfan #98: Thinkfan runs at 100% CPU usage, cannot be killed with CTRL+C

Saphieron, 2020-06-28:
Having cloned and build the repo following the README instructions with ccmake

The build configuration was 
CMAKE_BUILD_TYPE                 RelWithDebInfo                               
CMAKE_INSTALL_PREFIX             /usr/local                                   
DISABLE_BUGGER                   OFF                                          
DISABLE_EXCEPTION_CATCHING       OFF                                          
DISABLE_SYSLOG                   OFF                                          
USE_ATASMART                     OFF                                          
USE_NVML                         OFF                                          
USE_YAML                         ON

After it built successfully i installed it with `sudo make install`

I tried to testrun it with
`thinkfan -v -c /path/to/yamlFile`

From there it sits at 100% CPU usage, does not engage with the fan (which can be related to the yaml being wrong as well) and refuses to be terminated with Ctrl+C. Can only kill it via kill -9 <PID>.

Run on:

- Thinkpad X1 Yoga 3rd Gen
- Linux Mint 20 with Kernel 5.4.0-39-generic
- GNU Make 4.2.1
- g++ (Ubuntu 9.3.0-10ubuntu2) 9.3.0
- gcc (Ubuntu 9.3.0-10ubuntu2) 9.3.0

Having previously used the Ubuntu package containing thinkfan 0.9.3 i was able to control the fans with the old .conf format. So generally the fans do react.

vmatare, 2020-06-29:
Thanks for the report. But I need the config file you're using to be able to reproduce/fix this. You can just copy & paste it here.

vmatare, 2020-06-29:
In the meantime, you can try with one of the releases, maybe it behaves differently there.

Saphieron, 2020-06-29:
My current yaml only contains the following:
```
sensors:
  - hwmon: /sys/devices/platform/coretemp.0/hwmon
    indices: [3, 6, 4, 1, 5, 2]
  - tpacpi: /proc/acpi/ibm/thermal

fans:
  - tpacpi: /proc/acpi/ibm/fan

levels:
  - [0, 0, 55]
  - [1, 55, 61]
  - [2, 57, 62]
  - [3, 58, 63]
  - [4, 59, 65]
  - [5, 60, 66]
  - [6, 62, 67]
  - [7, 63, 255]

```
Not entirely sure that would even work. I will try to use this with a previous release build. 
(Mostly want to get the non-persistent hwmon paths to work with the new yaml functionality tbh.)

vmatare, 2021-08-29:
It's still not entirely clear what was causing this, but it could be the same as #145. So that would mean one of the indices `[3, 6, 4, 1, 5, 2]` didn't exist and with the most recent release thinkfan should no longer freeze and give a helpful error message in this situation. @Saphieron please test and report whether the latest updated on the master branch resolve the problem for you.

vmatare, 2022-02-06:
Closing due to no further feedback and because it's most likely fixed in v1.2.2 (#145)

closed 2022-02-06 by vmatare