vmatare/thinkfan #100: ERROR: /etc/thinkfan.conf: No fan specified in stepwise mapping.

ForresterAlex, 2020-07-10:
Installed thinkfan-git from AUR on manjaro/arch linux
thinkfan.conf is as follows:

`hwmon /sys/devices/platform/coretemp.0/hwmon/hwmon6/temp3_input
hwmon /sys/devices/platform/coretemp.0/hwmon/hwmon6/temp4_input
hwmon /sys/devices/platform/coretemp.0/hwmon/hwmon6/temp1_input
hwmon /sys/devices/platform/coretemp.0/hwmon/hwmon6/temp5_input
hwmon /sys/devices/platform/coretemp.0/hwmon/hwmon6/temp2_input

tp_fan /proc/acpi/ibm/fan

(0, 0, 50)

(1, 49, 55)

(2, 44, 60)

(3, 59, 65)

(5, 64, 70)

(7, 69, 78)

(64, 77, 32767)
`

sudo thinkfan -n returns:
`ERROR: /etc/thinkfan.conf:
No fan specified in stepwise mapping.
`

vmatare, 2020-07-12:
Sorry, this issue is specific to the legacy config syntax. Thanks for reporting it, but I also want to point out that the legacy config syntax will be deprecated in a future release. Of course this is a bug I'm going to fix, but I'd still like to encourage you to migrate to the new YAML config format. Take a look a the new `thinkfan.conf(5)` manpage and at the [example config](https://github.com/vmatare/thinkfan/blob/master/examples/thinkfan.yaml). If you find that it isn't helpful (enough) in converting your config, feel free to open another issue about that ;-)

closed 2020-07-12 by ForresterAlex

vmatare, 2020-07-12:
Should be fixed in the latest master now by commit #cd707386.