vmatare/thinkfan #105: ERROR: /etc/thinkfan.yaml: Invalid entry

TechieAndroid, 2020-08-11:
I've read every man page and wiki on this program and I can't get it to work. Here's my config:

```
sensors:
  - hwmon /sys/devices/pci0000:00/0000:00:18.3/hwmon/hwmon2/temp1_input
  - hwmon /sys/devices/pci0000:00/0000:00:18.3/hwmon/hwmon2/temp2_input
  - hwmon /sys/devices/pci0000:00/0000:00:01.7/0000:03:00.0/hwmon/hwmon0/temp3_input
  - hwmon /sys/devices/pci0000:00/0000:00:01.7/0000:03:00.0/hwmon/hwmon0/temp1_input
  - hwmon /sys/devices/pci0000:00/0000:00:01.7/0000:03:00.0/hwmon/hwmon0/temp2_input
  - hwmon /sys/devices/pci0000:00/0000:00:08.1/0000:04:00.0/hwmon/hwmon5/temp1_input

fans:
 -  tpacpi: /proc/acpi/ibm/fan

levels:
  - [0, 0,  41]
  - [1, 38, 51]
  - [2, 45, 56]
  - [3, 51, 61]
  - [4, 55, 64]
  - [5, 60, 66]
  - [6, 63, 68]
  - [7, 65, 74]
  - [127, 70, 3276]
```

OS: Arch Linux
CPU: Ryzen 7 3700U
GPU: Vega 10

Output of `cat /proc/acpi/ibm/fan`
```
status:         enabled
speed:          257
level:          1
commands:       level <level> (<level> is 0-7, auto, disengaged, full-speed)
commands:       enable, disable
commands:       watchdog <timeout> (<timeout> is 0 (off), 1-120 (seconds))
```
I don't understand the syntax of the yaml config file can someone tell me where I can go to learn how that's processed?

vmatare, 2020-08-12:
Thanks for the report. This looks like it could be issue #96, which should be fixed in the latest release. Which thinkfan version are you using?

TechieAndroid, 2020-08-17:
I just installed it the other day, it's version 1.2.1.

TechieAndroid, 2020-09-19:
Just wanted to provide an update on this issue.

I could never get thinkfan to work to control my fan, perhaps it doesn't support my model.
I just checked yesterday and I dont have and hwmon/fan or ibm/fan directory anymore, I'm not exactly sure what could've changed but I did a locate search and nothing came up for them, I also checked manually with ls /hwmon* and there wasn't anything related to fan control on my OS.

However, I did find one lone config file on my OS that I've never seen before located at /etc/amdgpu-fan.yml which contains a speed matix. After adjusting it though, nothing had changed even after a complete reboot.

In conclustion I don't think there's any simple way to control my laptop's fan, not unless I do a mod on the bios perhaps.

vmatare, 2020-09-20:
huh... what kind of laptop is it? You could try installing the `lm_sensors` package and running `sudo sensors-detect`.

TechieAndroid, 2020-09-20:
It's a Lenovo Flex 14 aka 81SS000BUS

Specs:

OS: Arch Linux

CPU: Ryzen 7 3700U 2.3GHz base / 4GHz boost / 1st gen Zen (Zen+)

GPU: Vega 10 1.4GHz

I already liquid metaled the CPU and I still don't quite reach the listed 4GHz, not even on one single core. I'd like to take control of the fan to let it run at max speed to try to keep the laptop under the thermal throttle limits.

I have had the lm_sensors package since long before I installed thinkfan. I'll run sensors-detect again though, just because ¯\_(ツ)_/¯ .

TechieAndroid, 2020-09-20:
I re-ran sensors-detect and then ran sensors, this was the output:

[image]

I also checked for any new fan related files or directories and there were none.

vmatare, 2020-09-20:
OK, that looks like there's no fan control driver available, at least not with your current distribution/kernel. One more thing you might try is [nbfc](https://github.com/hirschmann/nbfc). It's another fan control program that bypasses any hwmon drivers by writing directly into the registers of your embedded controller. Of course it requires some hardware-specific config to do that, but sometimes the community has already figured it out or they can help you do so. And btw, I'd be interested to hear how it worked out if you're going that route.

TechieAndroid, 2020-09-20:
Ah okay that explains why my system used to have some directories related to it a while back. Arch rolls out kernels pretty frequently so it probably changed recently and I wasn't aware of it.

Thanks for the heads up about that other program I'm definitely going to look into it. If I find out anything I'll let you know. Thanks for the help!

closed 2020-12-20 by vmatare