vmatare/thinkfan #106: Different fan curve on battery

saljs, 2020-08-12:
Not sure if this is too off topic, but a nice feature to add would be the ability to set a different fan curve when on battery. When I have my ThinkPad plugged in I want to set an aggressive curve to keep it from thermal throttling, but on battery I'd be okay with some throttling and lower fan speed to save power.

vmatare, 2020-08-12:
That is absolutely possible, but a little beyond the scope of thinkfan. What you want to do is set up a script that replaces the thinkfan config and then reloads the service. There's many ways to do that. [acpid](https://wiki.archlinux.org/index.php/Acpid) is one option, [udev](https://wiki.archlinux.org/index.php/Udev) is another. It's probably also possible with the help of systemd/upower but I'm not exactly sure how. If somebody was to contribute a nice little udev script, I'd definitely consider that for inclusion.

vmatare, 2020-08-12:
This might also be a good topic for a (first) wiki entry.

saljs, 2020-08-12:
Great suggestion, here's the script I wrote quick:

```
# Rule for when switching to battery
ACTION=="change", SUBSYSTEM=="power_supply", ATTR{type}=="Mains", ATTR{online}=="0", RUN+="/usr/bin/kill $( /run/thinkfan.pid); /usr/bin/thinkfan -c /etc/thinkfan/battery.conf"
# Rule for when switching to AC
ACTION=="change", SUBSYSTEM=="power_supply", ATTR{type}=="Mains", ATTR{online}=="1", RUN+="/usr/bin/kill $( /run/thinkfan.pid); /usr/bin/thinkfan -c /etc/thinkfan/charging.conf"
```

vmatare, 2020-08-26:
Reopening this as a reminder to add the udev script.

IvanSafonov, 2024-01-06:
I ended up with a new fan control [service](https://github.com/IvanSafonov/fanctl/blob/main/conf/thinkpad.yaml#L13-L25) :rofl: