vmatare/thinkfan #107: Ignoring existence of systemd under Mint

Bercik1337, 2020-08-20:
Hello

First of all - thanks for this project, I think I'm going to really appreciate it because I can't understand why fan control for Thinkpads sucks so bad.

Right to the point:
```
[bercik@neptune:~/nowy/thinkfan/build]$ uname -a
Linux neptune 5.4.0-42-generic #46-Ubuntu SMP Fri Jul 10 00:24:02 UTC 2020 x86
_64 x86_64 x86_64 GNU/Linux
[bercik@neptune:~/nowy/thinkfan/build]$ cat /etc/issue.net 
Linux Mint 20 Ulyana
[bercik@neptune:~/nowy/thinkfan/build]$ cmake ..
-- Checking for module 'openrc'
--   Package 'openrc', required by 'virtual:world', not found
-- Configuring done
-- Generating done
-- Build files have been written to: /home/bercik/nowy/thinkfan/build
[bercik@neptune:~/nowy/thinkfan/build]$ 
```
So I'm CLEARLY on systemd yet scripts wants (expects?) openrc to be installed? I managed to get down to only one "not found" in cmake but this stopped me for 2 days now. Or maybe I'm missing something/silly mistake?

Also throwing config here:
```
                                                    Page 1 of 1
 CMAKE_BUILD_TYPE                 RelWithDebInfo                             
 CMAKE_INSTALL_PREFIX             /usr/local                                 
 DISABLE_BUGGER                   OFF                                        
 DISABLE_EXCEPTION_CATCHING       OFF                                        
 DISABLE_SYSLOG                   OFF                                        
 USE_ATASMART                     OFF                                        
 USE_NVML                         ON                                         
 USE_YAML                         ON                                         

```

Thanks in advance for any help/

vmatare, 2020-08-20:
Hi, so what's the problem here? You don't have OpenRC, so it's not being found. Have you tried running `make`?

vmatare, 2020-08-20:
From a usability perspective I have to admit that the "required by ..." message is a little stupid in that it might give the impression that something is wrong. But it's not. The message is coming from cmake and I can't really get rid of it. You can report it as a cmake bug if you like ;-)

closed 2020-08-26 by vmatare