vmatare/thinkfan #109: "optional" keyword seems not to work

phrxmd, 2020-08-24:
Hi,

I have this in my `/etc/thinkfan.conf`:
```
sensors:
  # hwmon: Base path with name-based search
  # =======================================
  # Thinkfan will search under the given path for a hwmon driver that has a
  # file called "name" which contains the given name. This method should work
  # with all hwmon drivers and is robust against driver load order.
  - hwmon: /sys/class/hwmon
    name: coretemp
    indices: [1]
  - hwmon: /sys/class/hwmon
    name: acpitz
    indices: [1]
  - hwmon: /sys/class/hwmon
    name: pch_skylake
    indices: [1]
  - hwmon: /sys/class/hwmon
    name: iwlwifi_1
    indices: [1]
    optional: true
```

On bootup, systemd fails to bring up the thinkfand service with an error message related to the `iwlwifi` sensor, even though the "optional" keyword should supposedly prevent exactly that. Only when the system has fully booted up and I bring it up manually, it works:
```
-- Logs begin at Mon 2020-08-24 11:53:23 CEST, end at Mon 2020-08-24 13:53:23 CEST. --
Aug 24 11:53:26 calypso thinkfan[1345]: ERROR: read_temps: Failed to read temperature(s) from /sys/class/hwmon/hwmon7/temp1_input: No data available
Aug 24 11:56:01 calypso thinkfan[3914]: Daemon PID: 3924
Aug 24 11:56:01 calypso thinkfan[3924]: Temperatures(bias): 65(0), 55(0), 49(0), 47(0) -> level auto
```

phrxmd, 2020-09-05:
The system recognizes that a sensor is there (it detects a `hwmon` tree through the `name` file) - however then it fails to read an actual temperature from the sensor. The sensor is related to the built-in wifi card, which creates trouble when it overheats; maybe the temperature readout needs the wifi driver to be initialized.

vmatare, 2020-12-20:
There was an additional check on the config before thinkfan starts to operate for real. That check did not honour the `optional:` setting. I've now fixed that in the current master. Could you please try that out and see if it fixes your problem?

vmatare, 2021-06-23:
This should be fixed with the 1.2.2 release

closed 2021-06-23 by vmatare