vmatare/thinkfan #111: Unable to resume from sleep

ratcashdev, 2020-09-10:
Whenever I suspend my laptop (Lenovo X1 Extreme) running kernel 5.8.6, the thinkfan service failes to start.
```
~ >>> sudo systemctl status thinkfan                                                                                                                       
● thinkfan.service - simple and lightweight fan control program
     Loaded: loaded (/usr/lib/systemd/system/thinkfan.service; enabled; vendor preset: disabled)
    Drop-In: /etc/systemd/system/thinkfan.service.d
             └─override.conf
     Active: failed (Result: exit-code) since Thu 2020-09-10 11:43:48 CEST; 53s ago
    Process: 19462 ExecStart=/usr/bin/thinkfan $THINKFAN_ARGS -v (code=exited, status=0/SUCCESS)
   Main PID: 19463 (code=exited, status=1/FAILURE)

sep 10 11:42:59 rex-manjaro systemd[1]: Starting simple and lightweight fan control program...
sep 10 11:42:59 rex-manjaro thinkfan[19462]: /sys/devices/platform/thinkpad_hwmon/hwmon/hwmon7/pwm1: Restoring initial state: 2
sep 10 11:42:59 rex-manjaro thinkfan[19462]: Daemon PID: 19463
sep 10 11:42:59 rex-manjaro systemd[1]: Started simple and lightweight fan control program.
sep 10 11:42:59 rex-manjaro thinkfan[19463]: /sys/devices/platform/thinkpad_hwmon/hwmon/hwmon7/pwm1: Saved initial state: 2
sep 10 11:42:59 rex-manjaro thinkfan[19463]: Temperatures(bias): 47(0) -> level 0
sep 10 11:43:01 rex-manjaro thinkfan[19463]: Temperatures(bias): 51(8) -> level 64
sep 10 11:43:06 rex-manjaro thinkfan[19463]: Temperatures(bias): 44(0) -> level 32
// point of sleep/resume
sep 10 11:43:48 rex-manjaro systemd[1]: thinkfan.service: Main process exited, code=exited, status=1/FAILURE
sep 10 11:43:48 rex-manjaro systemd[1]: thinkfan.service: Failed with result 'exit-code'.
```

config:
```
sensors:
  - hwmon: /sys/devices/platform/thinkpad_hwmon/hwmon
    indices: [1]

fans:
  - hwmon: /sys/devices/platform/thinkpad_hwmon/hwmon
    indices: [1]

```

BohdanL, 2020-10-25:
I have upgraded to 1.2.1 in order to fix sleep issue and still having the issue:
```
říj 24 08:32:49 hermes thinkfan[448385]: Daemon PID: 448391
říj 24 08:32:49 hermes systemd[1]: Started Thinkfan, the minimalist fan control program.
říj 24 08:32:49 hermes thinkfan[448391]: Temperatures(bias): 41(0) -> level 0
říj 24 08:33:22 hermes thinkfan[448391]: Temperatures(bias): 52(0) -> level auto
říj 24 08:33:41 hermes thinkfan[448391]: Temperatures(bias): 49(0) -> level 0
říj 24 08:34:56 hermes thinkfan[448391]: Going to sleep: Will allow sensor read errors for the next 2 loops.
říj 24 09:30:38 hermes thinkfan[448391]: Received SIGUSR2: Re-initializing fan control.
říj 24 09:30:38 hermes thinkfan[448391]: ERROR: Lost sensor read_temps: Failed to read temperature(s) from /sys/class/hwmon/hwmon5/temp1_input: No such device or address
říj 24 09:30:38 hermes systemd[1]: thinkfan.service: Main process exited, code=exited, status=1/FAILURE
říj 24 09:30:38 hermes systemd[1]: thinkfan.service: Failed with result 'exit-code'.
```
sometimes exits even without error message, going to sleep is enough:
```
říj 25 08:03:21 hermes systemd[1]: Starting Thinkfan, the minimalist fan control program...
říj 25 08:03:21 hermes thinkfan[602723]: Daemon PID: 602733
říj 25 08:03:21 hermes systemd[1]: Started Thinkfan, the minimalist fan control program.
říj 25 08:03:21 hermes thinkfan[602733]: Temperatures(bias): 40(0) -> level 0
říj 25 08:07:42 hermes thinkfan[602733]: Going to sleep: Will allow sensor read errors for the next 2 loops.
říj 25 08:12:39 hermes systemd[1]: thinkfan.service: Main process exited, code=exited, status=1/FAILURE
říj 25 08:12:39 hermes systemd[1]: thinkfan.service: Failed with result 'exit-code'.
```
now is less resilient to sleep than 0.9.3

vmatare, 2020-12-20:
The weird thing about this is that there is no error message being logged. In all cases where thinkfan exits with a return code `1` (which means "expected, but unrecoverable error") there should be an error message.

What affected users could try is to disable exception handling so that thinkfan crashes instead of exiting. That will generate a coredump and may tell us more about the source of the problem. So you'd have to recompile with `DISABLE_EXCEPTION_CATCHING` set. So in the thinkfan "build" directory (assuming it's a subdirectory of thinkfan's main dir):
```bash
cmake -DDISABLE_EXCEPTION_CATCHING=ON ..
make -j4
```
Then run thinkfan in foreground in a terminal window (still in the "build" directory):
```
./thinkfan -n -v
```
Now make it crash by going to sleep and you should get a backtrace on the terminal window. Post it here and hopefully that will help me figure out the problem.

ratcashdev, 2020-12-20:
here is the stack trace:
```
sudo /usr/bin/thinkfan -n -v
/sys/devices/platform/thinkpad_hwmon/hwmon/hwmon7/pwm1: Saved initial state: 2.
Temperatures(bias): 42(0) -> level 0
ERROR: Unhandled thinkfan::SensorLost: Lost sensor read_temps: Failed to read temperature(s) from /sys/devices/platform/thinkpad_hwmon/hwmon/hwmon7/temp1_input: No such device or address.
errno = No such device or address.
Backtrace:
/usr/bin/thinkfan(+0x37415) [0x56136625c415]
/usr/bin/thinkfan(+0x20230) [0x561366245230]
/usr/lib/libstdc++.so.6(+0xa2d9a) [0x7fb43fd72d9a]
/usr/lib/libstdc++.so.6(+0xa2e07) [0x7fb43fd72e07]
/usr/lib/libstdc++.so.6(+0xa30ae) [0x7fb43fd730ae]
/usr/bin/thinkfan(void thinkfan::error<thinkfan::SensorLost, thinkfan::ExpectedError>(thinkfan::ExpectedError const&)+0x10f) [0x561366249c8f]
/usr/bin/thinkfan(+0x1b0d0) [0x5613662400d0]
/usr/bin/thinkfan(+0x1b260) [0x561366240260]
/usr/bin/thinkfan(thinkfan::run(thinkfan::Config const&)+0x1df) [0x561366248c7f]
/usr/bin/thinkfan(main+0x381) [0x561366246401]
/usr/lib/libc.so.6(__libc_start_main+0xf2) [0x7fb43fb13152]
/usr/bin/thinkfan(_start+0x2e) [0x561366246d4e]

zsh: abort      sudo /usr/bin/thinkfan -n -v
```

vmatare, 2020-12-20:
I'm sorry, I forgot you should also do a `Debug` build, so:
```bash
cmake -DDISABLE_EXCEPTION_CATCHING=ON -DCMAKE_BUILD_TYPE=Debug ..
make -j4
```
It also looks like you may not have debug symbols for libstdc++. If you open the core dump in a debugger, it should tell you what packages you need to install:
```bash
coredumpctl debug thinkfan
```

ratcashdev, 2020-12-20:
I am on Manjaro (Arch), so I am afraid no debug symbols...

vmatare, 2020-12-20:
> I am on Manjaro (Arch), so I am afraid no debug symbols...
At least for thinkfan you can get them with `-DCMAKE_BUILD_TYPE=Debug`. It's hard to imagine Manjaro really doesn't have any, but that's less important anyways.

ratcashdev, 2020-12-20:
Here's the output of the debug build:
```
/sys/devices/platform/thinkpad_hwmon/hwmon/hwmon7/pwm1: Saved initial state: 2.
Temperatures(bias): 47(0) -> level 0
Temperatures(bias): 46(0)
Temperatures(bias): 46(0)
Temperatures(bias): 47(0)
ERROR: Unhandled thinkfan::SensorLost: Lost sensor read_temps: Failed to read temperature(s) from /sys/devices/platform/thinkpad_hwmon/hwmon/hwmon7/temp1_input: No such device or address.
errno = No such device or address.
Backtrace:
/usr/bin/thinkfan(+0x9b785) [0x56075c57f785]
/usr/bin/thinkfan(thinkfan::handle_uncaught()+0x252) [0x56075c5800d2]
/usr/lib/libstdc++.so.6(+0xa2d9a) [0x7f5564bb4d9a]
/usr/lib/libstdc++.so.6(+0xa2e07) [0x7f5564bb4e07]
/usr/lib/libstdc++.so.6(+0xa30ae) [0x7f5564bb50ae]
/usr/bin/thinkfan(void thinkfan::error<thinkfan::SensorLost, thinkfan::ExpectedError>(thinkfan::ExpectedError const&)+0x5d) [0x56075c55e574]
/usr/bin/thinkfan(+0x763bd) [0x56075c55a3bd]
/usr/bin/thinkfan(+0x7659e) [0x56075c55a59e]
/usr/bin/thinkfan(thinkfan::run(thinkfan::Config const&)+0x23e) [0x56075c55a971]
/usr/bin/thinkfan(main+0x667) [0x56075c55d1d0]
/usr/lib/libc.so.6(__libc_start_main+0xf2) [0x7f5564955152]
/usr/bin/thinkfan(_start+0x2e) [0x56075c55a06e]

```

ratcashdev, 2020-12-21:
Did a small test of checking the file's existence while doing a suspend/resume during the duration of the loop:
```
while [ true ]; do ll /sys/devices/platform/thinkpad_hwmon/hwmon/hwmon7/temp1_input; sleep 1 ; done
-r--r--r-- 1 root root 4096 dec  8 20:57 /sys/devices/platform/thinkpad_hwmon/hwmon/hwmon7/temp1_input
-r--r--r-- 1 root root 4096 dec  8 20:57 /sys/devices/platform/thinkpad_hwmon/hwmon/hwmon7/temp1_input
-r--r--r-- 1 root root 4096 dec  8 20:57 /sys/devices/platform/thinkpad_hwmon/hwmon/hwmon7/temp1_input
-r--r--r-- 1 root root 4096 dec  8 20:57 /sys/devices/platform/thinkpad_hwmon/hwmon/hwmon7/temp1_input
-r--r--r-- 1 root root 4096 dec  8 20:57 /sys/devices/platform/thinkpad_hwmon/hwmon/hwmon7/temp1_input
-r--r--r-- 1 root root 4096 dec  8 20:57 /sys/devices/platform/thinkpad_hwmon/hwmon/hwmon7/temp1_input
-r--r--r-- 1 root root 4096 dec  8 20:57 /sys/devices/platform/thinkpad_hwmon/hwmon/hwmon7/temp1_input
-r--r--r-- 1 root root 4096 dec  8 20:57 /sys/devices/platform/thinkpad_hwmon/hwmon/hwmon7/temp1_input
```
So far so good, the file is there even after suspend/resume.
The next was to write out the contents of the file:
```
while [ true ]; do cat /sys/devices/platform/thinkpad_hwmon/hwmon/hwmon7/temp1_input; sleep 1 ; done
40000
40000
40000
40000
cat: /sys/devices/platform/thinkpad_hwmon/hwmon/hwmon7/temp1_input: No such device or address
cat: /sys/devices/platform/thinkpad_hwmon/hwmon/hwmon7/temp1_input: No such device or address
cat: /sys/devices/platform/thinkpad_hwmon/hwmon/hwmon7/temp1_input: No such device or address
cat: /sys/devices/platform/thinkpad_hwmon/hwmon/hwmon7/temp1_input: No such device or address
42000
42000
42000
```

Apparently shortly after resume the file is there, but it's not possible to open it.

ratcashdev, 2021-01-01:
same on Linux 5.10.2

ratcashdev, 2021-02-03:
@vmatare any idea what to do with this?

ratcashdev, 2021-03-22:
testing if adding 
```
RestartSec=5
Restart=on-failure
```
to the systemd `[Unit]` section helps

skligys, 2021-04-14:
Would adding
```
ExecStartPre=/usr/bin/sleep 5
```
to `thinkfan-wakeup.service` work?

ratcashdev, 2021-04-14:
Unfortunately it does not help. Journal says:
```
apr 14 21:15:55 mypc systemd[1]: Started Thinkfan, the minimalist fan control program.
apr 14 21:15:55 mypc thinkfan[35955]: Temperatures(bias): 36(0) -> level 0
apr 14 21:16:02 mypc thinkfan[35955]: Going to sleep: Will allow sensor read errors for the next 2 loops.
apr 14 21:16:19 mypc systemd[1]: thinkfan.service: Main process exited, code=exited, status=1/FAILURE
apr 14 21:16:19 mypc systemd[1]: thinkfan.service: Failed with result 'exit-code'.
```

So apparently those 2 cycles may finish before the laptop indeed goes to sleep. By the time the wakeup service is called, there's nothing running any more.

ratcashdev, 2021-04-14:
solved using the (rather brute) approach, by adding the restart condition to the `thinkfan.service` itself. It now says:
```
[Service]
Type=forking
ExecStart=/usr/bin/thinkfan $THINKFAN_ARGS
PIDFile=/run/thinkfan.pid
ExecReload=/bin/kill -HUP $MAINPID
**RestartSec=5**
**Restart=on-failure**
```

The result:

```
apr 14 21:46:39 mypc thinkfan[38243]: Temperatures(bias): 36(0) -> level 0
apr 14 21:46:45 mypc thinkfan[38243]: Going to sleep: Will allow sensor read errors for the next 2 loops.
apr 14 21:47:00 mypc systemd[1]: thinkfan.service: Main process exited, code=exited, status=1/FAILURE
apr 14 21:47:00 mypc systemd[1]: thinkfan.service: Failed with result 'exit-code'.
apr 14 21:47:05 mypc systemd[1]: thinkfan.service: Scheduled restart job, restart counter is at 1.
apr 14 21:47:05 mypc systemd[1]: Stopped Thinkfan, the minimalist fan control program.
apr 14 21:47:05 mypc systemd[1]: Starting Thinkfan, the minimalist fan control program...
apr 14 21:47:05 mypc thinkfan[38917]: Daemon PID: 38919
apr 14 21:47:05 mypc systemd[1]: Started Thinkfan, the minimalist fan control program.
apr 14 21:47:05 mypc thinkfan[38919]: Temperatures(bias): 42(0) -> level 0
apr 14 21:47:05 mypc thinkfan[38919]: Received SIGUSR2: Re-initializing fan control.
apr 14 21:47:05 mypc thinkfan[38919]: Temperatures(bias): 42(0) -> level 0
```

vmatare, 2021-04-15:
I gave the relevant code another review and I think I found a logic mixup that would explain the behavior you're seeing. Could anyone affected by this please test with the current master again and report back?

ratcashdev, 2021-04-15:
Looks good, doesn't it?

```
sudo ./thinkfan -n -v                                                                                                                                             1   master 
/sys/devices/platform/thinkpad_hwmon/hwmon/hwmon6/pwm1: Saved initial state: 2.
Temperatures(bias): 45(0) -> Fans: 32
Going to sleep: Will allow sensor read errors for the next 2 loops.
Lost sensor read_temps: Failed to read temperature(s) from /sys/devices/platform/thinkpad_hwmon/hwmon/hwmon6/temp1_input: No such device or address
Temperatures(bias): -128(0) -> Fans: 0
Received SIGUSR2: Re-initializing fan control.
Lost sensor read_temps: Failed to read temperature(s) from /sys/devices/platform/thinkpad_hwmon/hwmon/hwmon6/temp1_input: No such device or address
Temperatures(bias): -128(0) -> Fans: 0
Temperatures(bias): 47(262) -> Fans: 255
Temperatures(bias): 46(0) -> Fans: 32
```

vmatare, 2021-04-16:
Awesome! Thanks @ratcashdev for reporting back. I'll fix #123 as well and then make a bugfix release.

closed 2021-04-16 by vmatare