vmatare/thinkfan #114: When on level "0": Unexpected short fan activation.

kolibril13, 2020-10-12:
I have a Thinkpad T495 Ubuntu 20.4 and want to use Thinkfan, as I don't want to have fan noises when e.g. watching videos.
I installed `thinkfan 0.9.1`, and it workes more or less, except that about every two minutes, the fan is starting to spin but also stopping immediately (starting+stopping takes maybe 3 seconds) at low temperatures (~45°C), where I would expect level 0.
My file `/etc/thinkfan.conf` looks like this:
```console
hwmon /sys/devices/pci0000:00/0000:00:18.3/hwmon/hwmon2/temp1_input

(0,	0,	85)
(1,	85,	86)
(2,	86,	87)
(3,	87,	88)
(4,	88,	89)
(5,	89,	90)
(6,	90,	91)
(7,	91,	32767)
```
Can someone maybe give me any advice?
My theory is that the Ubuntu internal and the think fan scripts to control the fan are interfering somehow.
 Thanks a lot!

vmatare, 2020-10-12:
Hi, that sounds like a bug that has been fixed for a very, very long time now. I think it should already be fixed in the [0.9.3 release](https://github.com/vmatare/thinkfan/releases/tag/0.9.3), and that would be from 2016. Since then thinkfan has undergone a complete redesign and we're currently at [1.2.1](https://github.com/vmatare/thinkfan/releases/tag/1.2.1).

Someone (preferrably a Ubuntu user, cough cough) should ping the package maintainer and tell them to update ;-)

vmatare, 2020-10-12:
oh, and are you actually serious about keeping the fan off until your temperature has reached 85°C? I don't know what you're (not) cooling there, but 85°C sounds awfully hot for any device...

kolibril13, 2020-10-13:
Thanks a lot, @vmatare !
When googling thinkfan, there are really few hints that there is a newer version.
Maybe one can also note in the latest release in the wiki, https://github.com/vmatare/thinkfan/wiki , as there are already the other releases mentioned.
I can ping the package maintainer, can you tell me where to find them?
Is there an email address, or a website?
At the moment, when one types `sudo apt-get install thinkfan` it will install thinkfan 0.9.1

vmatare, 2020-10-13:
> However, I still have the problem of unexpected short fan activation.
> I just measured the time, exactly every 112 seconds the fan starts to spin, and stops then again, which is also quite annoying.
> Any ideas of how I can suppress that?

Yes, upgrade at least to 0.9.3. Even Debian have it in their stable repos: https://packages.debian.org/stable/thinkfan
In fact, you might be able to just install the debian package.

And then, to wake up the Ubuntu people you should report it as a Ubuntu bug: https://bugs.launchpad.net/ubuntu

vmatare, 2020-10-13:
Sorry, I just saw that thinkfan 0.9.3 incorrectly reports its version number as 0.9.1. So it's likely that your package database says 0.9.3, while `thinkfan -h` will say 0.9.1. In that case, you probably do in fact have 0.9.3.
But as I said, all of the 0.x versions are ANCIENT and I don't support them any more. I might think of publishing a Debian/Ubuntu package myself but right now I don't have the time. So for the time being I'd ask you to uninstall the Ubuntu package and manually upgrade to thinkfan 1.2.1.

Oh, and please don't downgrade, I don't believe that will fix your problem.

If you actually observed this problem 1.2 or 1.2.1, it might be something I need to look at. So are you saying the behavior is exactly the same as with the old 0.9 version?

kolibril13, 2020-10-13:
> If you actually observed this problem with 1.2 or 1.2.1, it might be something I need to look at. So are you saying the behaviour is exactly the same as with the old 0.9 version?

`sudo thinkfan -h` gives me the following output:
```console
thinkfan 1.2: A minimalist fan control program
```

> So for the time being I'd ask you to uninstall the Ubuntu package and manually upgrade to thinkfan 1.2.1.

This is exactly what I did, and the problem is still there.
If you want to have a look at that, that would be really great and I would appreciate this a lot!
Feel free to contact me if you want to have support.

vmatare, 2020-10-14:
Which command line options are you using when you launch thinkfan? In particular, are you using the `-p` or `-s` options?

kolibril13, 2020-10-26:
Hi @vmatare , sorry, I somehow overlooked your last message. Tanks again for helping me with this.
I tried now both options, `-p` or `-s`, both times still the same behaviour with the short fan activation.
Here my console output:
```console
$ sudo thinkfan -s 10
/proc/acpi/ibm/fan: Restoring initial state: auto.
Daemon PID: 30651
````
```console
$ sudo thinkfan -p
Disengaging the fan controller for 5 seconds every 0.500000 seconds
/proc/acpi/ibm/fan: Restoring initial state: auto.
Daemon PID: 30207
```

kolibril13, 2020-11-13:
@vmatare : any ideas maybe?
Or will switching from ubuntu to Debian (which I already thought of in another context) solve this problem?

mhhayashi, 2021-01-10:
I have the same issue on my Thinkpad T14s (AMD) running Arch.

Whenever /proc/acpi/ibm/fan is written to, the fan will spin for half a second. Since thinkfan writes to this file every 120 seconds, the fan spins every 120 seconds. 

Changing the number on line 84 of drivers.cpp changes how often the fan spins
https://github.com/vmatare/thinkfan/blob/eab6f5bf53fc93b7677f541cb5dc92b13aa4243e/src/drivers.cpp#L84

buck2202, 2022-01-08:
@vmatare I'm not personally affected by this (the fan in my T430 never gets to zero...), but came here via a reddit post about this issue. The user claims that writing `watchdog xx` periodically (instead of re-setting the current speed) is enough to both keep the watchdog happy, and doesn't result in momentary fan spin up if the current level is zero. AMD T14
https://www.reddit.com/r/thinkpad/comments/rw2xur/is_fan_control_supposed_to_be_persistent_with/

Thought it was worth mentioning, so maybe someone affected can test/confirm that here.

pennae, 2022-07-15:
affected and can confirm. then AMD T14 Gen2 spins the fan up very briefly even when turning it off. this patch is sufficient to stop the 2-minutely spinup while preventing watchdog reset (on 1.3.1)

```patch
diff --git a/src/fans.cpp b/src/fans.cpp
index 56a15d7..08c9b4a 100644
--- a/src/fans.cpp
+++ b/src/fans.cpp
@@ -123,7 +123,11 @@ void TpFanDriver::ping_watchdog_and_depulse(const Level &level)
 	}
 	else if (last_watchdog_ping_ + watchdog_ - sleeptime <= std::chrono::system_clock::now()) {
 		log(TF_DBG) << "Watchdog ping" << flush;
-		set_speed(level);
+		std::fstream f(path_);
+		if (!(f.is_open() && f.good()))
+			throw IOerror(MSG_FAN_INIT(path_), errno);
+		if (!(f << "watchdog " << watchdog_.count() << std::flush))
+			throw IOerror(MSG_FAN_INIT(path_), errno);
 	}
 }
```

peter-stoll, 2024-02-29:
I'm affected and can confirm that the patch works with 1.3.1 on a T495 (actually a T490 with AMD CPU).
The patch is not in the current HEAD. Is there any reason why the patch has not found it's way into HEAD after 1 1/2 years?
Or is it just, that nobody found the time to do create a pull request or whatever?

tova-cup, 2024-09-01:
> I'm affected and can confirm that the patch works with 1.3.1 on a T495 (actually a T490 with AMD CPU). The patch is not in the current HEAD. Is there any reason why the patch has not found it's way into HEAD after 1 1/2 years? Or is it just, that nobody found the time to create a pull request or whatever?

@peter-stoll, I stumbled upon this thread while facing the issue myself and I took the matter into my own hands.

A pull request (see above) was submitted ; it is pending merging by vmatare.