vmatare/thinkfan #116: Very confused about choosing sensors

Jackojc, 2020-11-16:
Hi, I'm having issues configuring my sensors.

I notice that my paths for sensors in `/sys/class/hwmon` are not consistent across reboots.

My `/sys/class/hwmon` directory has 5 folders named `hwmon0` through `hwmon5`, which one should I be looking at to get the proper temperature readings?

Additionally, `lm_sensors` lists 8 different sensors, `temp1_input` through `temp8_input`, most of them seem to just read 0c though.

The output after running `sensors` looks something like this. I'm on an x240.
```
thinkpad-isa-0000
Adapter: ISA adapter
fan1:        3064 RPM
temp1:        +45.0°C
temp2:         +0.0°C
temp3:         +0.0°C
temp4:         +0.0°C
temp5:         +0.0°C
temp6:         +0.0°C
temp7:         +0.0°C
temp8:         +0.0°C

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +46.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +46.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +45.0°C  (high = +100.0°C, crit = +100.0°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +45.0°C  (crit = +200.0°C)
```

I've tried using this:
```
sensors:
  - hwmon: /sys/class/hwmon
    name: temp1_input

  - hwmon: /sys/class/hwmon
    name: temp2_input

  - hwmon: /sys/class/hwmon
    name: temp3_input
```
but thinkfan seems to complain that those files don't exist. As I understand it, this "fuzzy match" thing should look at all the folders in `/sys/class/hwmon` and find which folders have files named `temp*_input`? I've also tried using:
```
sensors:
  - hwmon: /sys/class/hwmon
    indices: [1, 2, 3]
```
But this also doesn't seem to work and will complain across reboots I'm sure. I don't really understand what indices does either, is it supposed to treat, say, the output of `ls -1` as an array and it just picks out the sensors at those indexes?

Just really confused about how this is supposed to work. Any help would be much appreciated, thanks.

vmatare, 2020-12-20:
Well, did you try reading the config manpage: `man thinkfan.conf`? An [example config](https://github.com/vmatare/thinkfan/blob/master/examples/thinkfan.yaml) has also recently been added.

Together, these two should tell you all you need to know. If they don't you could help me to improve them by telling me what exactly you read there and what did or did not understand from it.

Sorry if I'm reluctant to answer these issues directly, but the documentation should explain all these things and if it doesn't, I need your help to improve it accordingly ;-)

vmatare, 2021-06-28:
Assuming that this has indeed been answered by the documentation.

closed 2021-06-28 by vmatare