vmatare/thinkfan #118: Autostart fails, and restart is needed because hwmon is not found

lakotamm, 2020-12-15:
Hello,
For the last few days, I have been experiencing issues with thinkfan failing to start, because it cannot find hwmon.

OS: Manjaro Linux x86_64 - testing branch
Host: ThinkPad T490
Kernel: 5.9.12-1-MANJARO 
CPU: Intel i7-8565U 

<pre><font color="#CC0000"><b>●</b></font> thinkfan.service - simple and lightweight fan control program
     Loaded: loaded (/usr/lib/systemd/system/thinkfan.service; enabled; vendor preset: disabled)
    Drop-In: /etc/systemd/system/thinkfan.service.d
             └─override.conf
     Active: <font color="#CC0000"><b>failed</b></font> (Result: exit-code) since Tue 2020-12-15 21:58:42 CET; 1h 25min ago
    Process: 520 ExecStart=/usr/bin/thinkfan $THINKFAN_ARGS <font color="#CC0000"><b>(code=exited, status=1/FAILURE)</b></font>

Dec 15 21:58:42 miroslav-20n2cto1ww systemd[1]: Starting simple and lightweight fan control program...
Dec 15 21:58:42 miroslav-20n2cto1ww thinkfan[520]: <font color="#CC0000"><b>ERROR: Error scanning /sys/devices/platform/coretemp.0/hwmon: No such file or directory</b></font>
Dec 15 21:58:42 miroslav-20n2cto1ww systemd[1]: <b>thinkfan.service: Control process exited, code=exited, status=1/FAILURE</b>
Dec 15 21:58:42 miroslav-20n2cto1ww systemd[1]: <font color="#D7D75F"><b>thinkfan.service: Failed with result &apos;exit-code&apos;.</b></font>
Dec 15 21:58:42 miroslav-20n2cto1ww systemd[1]: <font color="#CC0000"><b>Failed to start simple and lightweight fan control program.</b></font>
</pre>

thinkfan.yaml:

`sensors:
  - hwmon: /sys/devices/platform/coretemp.0/hwmon
    indices: [2, 3, 4, 5]

levels:
  - [0, 0 , 55]
  - [1, 50, 60]
  - [2, 55, 65]
  - [3, 60, 70]
  - [4, 65, 75]
  - [5, 70, 80]
  - [6, 75, 85]
  - [7, 80, 32767]
`
Since restarting the service fixes the issue, I assume that hwmon is initialized after thinkfan. Is this a more widespread issue or just me having messed up configuration?

vmatare, 2020-12-15:
I've made a small change, can you uninstall thinkfan and re-try with the latest master built from github?

lakotamm, 2020-12-16:
Unfortunately no improvement. I double-checked that got rid of the old version from the AUR + that I pulled the latest master. I tried it compiled for both release and debug.

<pre><font color="#CC0000"><b>●</b></font> thinkfan.service - Thinkfan, the minimalist fan control program
     Loaded: loaded (/usr/local/lib/systemd/system/thinkfan.service; enabled; vendor preset: disabled)
    Drop-In: /etc/systemd/system/thinkfan.service.d
             └─default.conf, override.conf
     Active: <font color="#CC0000"><b>failed</b></font> (Result: exit-code) since Wed 2020-12-16 01:02:42 CET; 43s ago
    Process: 490 ExecStart=/usr/local/sbin/thinkfan $THINKFAN_ARGS <font color="#CC0000"><b>(code=exited, status=1/FAILURE)</b></font>

Dec 16 01:02:42 miroslav-20n2cto1ww systemd[1]: Starting Thinkfan, the minimalist fan control program...
Dec 16 01:02:42 miroslav-20n2cto1ww thinkfan[490]: <font color="#CC0000"><b>ERROR: Error scanning /sys/devices/platform/coretemp.0/hwmon: No s</b></font><span style="background-color:#DCE9EB"><font color="#1B2224">&gt;</font></span>
Dec 16 01:02:42 miroslav-20n2cto1ww systemd[1]: <b>thinkfan.service: Control process exited, code=exited, status=1/FAILU</b><span style="background-color:#DCE9EB"><font color="#1B2224">&gt;</font></span>
Dec 16 01:02:42 miroslav-20n2cto1ww systemd[1]: <font color="#D7D75F"><b>thinkfan.service: Failed with result &apos;exit-code&apos;.</b></font>
Dec 16 01:02:42 miroslav-20n2cto1ww systemd[1]: <font color="#CC0000"><b>Failed to start Thinkfan, the minimalist fan control program.</b></font></pre>

vmatare, 2020-12-16:
Can you give me the output of `systemctl list-dependencies thinkfan` please?

lakotamm, 2020-12-16:
Here it is:
<pre> &gt;<font color="#06989A">&gt;</font><font color="#CC0000"><b>&gt;</b></font> <font color="#4E9A06">systemctl</font> list-dependencies <u style="text-decoration-style:single">thinkfan</u>                                                      <font color="#CC0000"> [130]</font>
thinkfan.service
<font color="#4E9A06"><b>●</b></font> ├─system.slice
<font color="#4E9A06"><b>●</b></font> └─sysinit.target
<font color="#4E9A06"><b>●</b></font>   ├─dev-hugepages.mount
<font color="#4E9A06"><b>●</b></font>   ├─dev-mqueue.mount
<font color="#4E9A06"><b>●</b></font>   ├─kmod-static-nodes.service
●   ├─ldconfig.service
<font color="#4E9A06"><b>●</b></font>   ├─lvm2-lvmetad.socket
<font color="#4E9A06"><b>●</b></font>   ├─lvm2-lvmpolld.socket
<font color="#4E9A06"><b>●</b></font>   ├─lvm2-monitor.service
<font color="#4E9A06"><b>●</b></font>   ├─proc-sys-fs-binfmt_misc.automount
<font color="#4E9A06"><b>●</b></font>   ├─sys-fs-fuse-connections.mount
<font color="#4E9A06"><b>●</b></font>   ├─sys-kernel-config.mount
<font color="#4E9A06"><b>●</b></font>   ├─sys-kernel-debug.mount
<font color="#4E9A06"><b>●</b></font>   ├─sys-kernel-tracing.mount
<font color="#4E9A06"><b>●</b></font>   ├─systemd-ask-password-console.path
<font color="#4E9A06"><b>●</b></font>   ├─systemd-binfmt.service
●   ├─systemd-boot-system-token.service
●   ├─systemd-firstboot.service
●   ├─systemd-hwdb-update.service
●   ├─systemd-journal-catalog-update.service
<font color="#4E9A06"><b>●</b></font>   ├─systemd-journal-flush.service
<font color="#4E9A06"><b>●</b></font>   ├─systemd-journald.service
●   ├─systemd-machine-id-commit.service
<font color="#4E9A06"><b>●</b></font>   ├─systemd-modules-load.service
<font color="#4E9A06"><b>●</b></font>   ├─systemd-random-seed.service
●   ├─systemd-repart.service
<font color="#4E9A06"><b>●</b></font>   ├─systemd-sysctl.service
●   ├─systemd-sysusers.service
<font color="#4E9A06"><b>●</b></font>   ├─systemd-timesyncd.service
<font color="#4E9A06"><b>●</b></font>   ├─systemd-tmpfiles-setup-dev.service
<font color="#4E9A06"><b>●</b></font>   ├─systemd-tmpfiles-setup.service
<font color="#4E9A06"><b>●</b></font>   ├─systemd-udev-trigger.service
<font color="#4E9A06"><b>●</b></font>   ├─systemd-udevd.service
●   ├─systemd-update-done.service
<font color="#4E9A06"><b>●</b></font>   ├─systemd-update-utmp.service
<font color="#4E9A06"><b>●</b></font>   ├─cryptsetup.target
<font color="#4E9A06"><b>●</b></font>   ├─local-fs.target
<font color="#4E9A06"><b>●</b></font>   │ ├─-.mount
<font color="#4E9A06"><b>●</b></font>   │ ├─boot-efi.mount
●   │ ├─systemd-fsck-root.service
<font color="#4E9A06"><b>●</b></font>   │ ├─systemd-remount-fs.service
<font color="#4E9A06"><b>●</b></font>   │ └─tmp.mount
<font color="#4E9A06"><b>●</b></font>   └─swap.target
</pre>

vmatare, 2020-12-16:
OK, so I was looking for a dependency on `systemd-modules-load.service`. I added it to the thinkfan service file, but it also seems to be part of `sysinit.target` on which thinkfan has always depended. So thinkfan **should** start only after all modules have been initialized. Assuming systemd dependency handling works correctly, the only explanation would be that a module takes very long to initialize its sysfs files after being loaded.

I was always under the impression that an application like thinkfan shouldn't be responsible for this problem. This is dependency management and therefore it's the job of the system's service manager. So technically, this should be taken up with either systemd or the kernel itself.

However for the sake of practicality, I might implement a workaround thinkfan. For the time being, affected users should find some other way to delay thinkfan startup. For examle:

Do `systemctl edit thinkfan` and add the following:
```ini
[Unit]
After=graphical.target
```

lakotamm, 2020-12-16:
If you want, I can try to make a bug report, however, my knowledge of systemd is very limited, so I might not be able to answer their questions.

Could it be that it has something to do with optimus-manager? As far as I remember it could become broken after the last update of the package. I could try to investigate.

I tried your suggested workaround, and it resulted in a service being dead (inactive), not starting up.
`[Unit]
After=graphical.target`

so I tried instead `basic.target` and that just meant that the issue continued. Using `gdm.service` fixed the issue.

Printed `sudo systemd-analyze critical-chain                   `                                          
```
graphical.target @2.120s
└─multi-user.target @2.120s
  └─thinkfan.service @2.060s +60ms
    └─gdm.service @2.052s +6ms
      └─optimus-manager.service @1.051s +999ms
        └─basic.target @1.043s
          └─sockets.target @1.043s
            └─snapd.socket @1.043s +534us
              └─sysinit.target @1.041s
                └─systemd-backlight@backlight:intel_backlight.service @1.646s +2.195s
                  └─system-systemd\x2dbacklight.slice @1.445s
                    └─system.slice @231ms
                      └─-.slice @231ms
```

vmatare, 2020-12-16:
Anyways, thanks for this helpful report. Since the error says `Error scanning /sys/devices/platform/coretemp.0/hwmon`, I would think that it's related to the `coretemp` module (for the CPU). It could also be an issue with the module where it completes its initialization procedure before having set up all sysfs symlinks.

Oh, and there's also the possibility that something else is actually un-loading and re-loading modules during bootup. If that happens during thinkfan startup, there's bound to be trouble.

So I think the precise cause of the issue should be investigated more closely before reporting bugs anywhere else.

phrxmd, 2020-12-20:
I have had what I think is the same issue on OpenSUSE for a few days. I guess it appeared after a recent kernel update to 5.9.14, but it might have crept in earlier. I have `coretemp`, `acpitz` and `pch-skylake` hwmons in my `/etc/thinkfan.conf`. It systematically fails on bootup with error messages like this:

```
> sudo systemctl status thinkfan
● thinkfan.service - Thinkfan, the minimalist fan control program
     Loaded: loaded (/usr/lib/systemd/system/thinkfan.service; enabled; vendor preset: disabled)
    Drop-In: /etc/systemd/system/thinkfan.service.d
             └─override.conf
     Active: failed (Result: exit-code) since Sun 2020-12-20 13:21:04 CET; 43s ago
    Process: 890 ExecStart=/usr/sbin/thinkfan $THINKFAN_ARGS (code=exited, status=1/FAILURE)

Dez 20 13:21:04 calypso systemd[1]: Starting Thinkfan, the minimalist fan control program...
Dez 20 13:21:04 calypso thinkfan[890]: ERROR: /etc/thinkfan.conf:52:
                                           name: coretemp
                                                 ^
                                       Could not find a hwmon with this name.
Dez 20 13:21:04 calypso systemd[1]: thinkfan.service: Control process exited, code=exited, status=1/FAILURE
Dez 20 13:21:04 calypso systemd[1]: thinkfan.service: Failed with result 'exit-code'.
Dez 20 13:21:04 calypso systemd[1]: Failed to start Thinkfan, the minimalist fan control program.
```

So I think there is a dependency problem, but I haven't figured out the correct load order yet.

vmatare, 2020-12-20:
Just thought of another thing: The `coretemp` module is probably loaded on-demand and I'm not sure exactly how/when that is triggered. So you could try loading it explicitly at bootup via `systemd-modules-load.service`:
```bash
echo coretemp > /etc/modules-load.d/coretemp.conf
```

TheFineBroccoli, 2021-05-04:
I had the same issue, solved it by making the service file restart-able on failure:  

```ini
Restart=on-failure
RestartSec=5s
```

ceigh, 2021-08-06:
I just delay it as tmp fix:
```ini
[Service]
ExecStartPre=/bin/sleep 10
```

lakotamm, 2021-10-09:
I just encountered the same issue on Fedora Silverblue 35. Adding 10s startup delay is still an easy fix.

vmatare, 2022-03-29:
Fix is currently in development at https://github.com/vmatare/thinkfan/tree/init_retries. For now, the branch simply makes the `optional` keyword work at startup, too, providing a workaround for this problem. Soon I'll add another option `max_errors` which will allow a given number of consecutive init or read errors to be ignored on a given sensor. That'll become the designated way to handle the startup issues in the 2.0 release.

vmatare, 2022-08-12:
@lakotamm @ceigh @broccoli5 @phrxmd, this should be fixed in master now with a new config setting: https://github.com/vmatare/thinkfan/issues/189#issuecomment-1196011925

closed 2022-08-13 by vmatare