vmatare/thinkfan #119: More config file examples

SirHackselot, 2021-01-16:
Thank you so much for the YAML functionality. Finally, I don't have to be constantly changing the thinkfan.conf file!

I think more config file examples would be useful, or maybe a "discussion" about config files for specific machines.

This is working on my T520:

sensors:
  - hwmon: /sys/class/hwmon  
    name: coretemp
    indices: [1, 2, 3, 4, 5]
  - hwmon: /sys/class/hwmon                                         
    name: acpitz
    indices: [1]

fans:
  - tpacpi: /proc/acpi/ibm/fan

levels:
  - [0,      0,     55]
  - [1,     48,     60]
  - [2,     50,     61]
  - [3,     52,     63]
  - [4,     56,     67]
  - [5,     59,     70]
  - [7,     63,     80]
  - [127,   75,     32767]

dkowis, 2021-07-12:
A wiki site where folks can publish their configs or something like that would be stupid useful.

vmatare, 2021-07-25:
I've added a specific category for example configs under Discussions and added some additional info to your contribution. Thanks for the idea, let's see how this goes ;-)

closed 2021-07-25 by vmatare