vmatare/thinkfan #122: Using rolling average instead auf direct sensor data

Carabas90, 2021-02-04:
Hi there,
at first: Thanks for this amazing program. The standard fan-setting were running my ThinkPad at 70-80 Degrees + all the time, even under relatively light loads. 

I have just one problem the sensor data i get from __tp_thermal /proc/acpi/ibm/thermal__ has a lot of short spikes to 80-90 Degrees in it which lead to the fanspeed ramping up to 5-6 for a second or two, than slowing down again.

This is really annoying an can be to good for the fans in the long term.

That's why I was wondering whether it would be possible to add a option to use a rolling average of the sensor data over a set timeframe instead of the raw sensor data.

I already tried reading the Code and maybe implementing this myself, but I know way to less about C++ (almost nothing to honest) and CPU-Cooling to tackle that safely.

vmatare, 2021-02-04:
We do have something similar with the `-b` commandline option. It's not a moving average, but it can help even out those short temperature spikes. It can be used both to "exaggerate" rising temperatures (with positive arguments) and to smoothen the temperature curve (with negative arguments). The exact formula used to calculate the resulting control input is in the manpage `thinkfan (1)`.

Carabas90, 2021-02-04:
Thank you that improved it a lot. 
Sorry for bothering you.

closed 2021-02-04 by Carabas90