vmatare/thinkfan #123: fan at full-speed after resume from sleep

rnhmjoj, 2021-02-04:
As soon as the system resumes from sleep (eg. close and reopen the lid), thinkfan set the fan at full speed.
The problem seems to be the temperature jumps from -128 to 45. Is there anything I can do to avoid this?

Log:
```
12:10:27 systemd[1]:     Starting Notify thinkfan of imminent sleep...
12:10:27 thinkfan[9307]: Going to sleep: Will allow sensor read errors for the next 2 loops.
12:10:28 systemd[1]:     thinkfan-sleep.service: Succeeded.
12:10:28 systemd[1]:     Finished Notify thinkfan of imminent sleep.
12:10:37 systemd[1]:     Starting Reload thinkfan after waking up from suspend...
12:10:37 thinkfan[9307]: Received SIGUSR2: Re-initializing fan control.
12:10:37 systemd[1]:     thinkfan-wakeup.service: Succeeded.
12:10:37 systemd[1]:     Finished Reload thinkfan after waking up from suspend.
12:10:37 thinkfan[9307]: Temperatures(bias): -128(0), 0(0), 0(0), 0(0), 0(0), 0(0), 0(0), 0(0) -> level 0
12:10:42 thinkfan[9307]: Temperatures(bias): 45(259), 0(0), 0(0), 0(0), 0(0), 0(0), 0(0), 0(0) -> level 7
12:10:57 thinkfan[9307]: Temperatures(bias): 45(0), 0(0), 0(0), 0(0), 0(0), 0(0), 0(0), 0(0) -> level 0
```
config:
```
fans:
  - tpacpi: /proc/acpi/ibm/fan
levels:
  - [0,  0, 58]
  - [1, 53, 65]
  - [2, 55, 66]
  - [3, 57, 68]
  - [4, 61, 70]
  - [5, 64, 71]
  - [7, 68, 32767]
sensors:
  - tpacpi: /proc/acpi/ibm/thermal
```

rnhmjoj, 2021-02-04:
From [thinkpad-acpi](https://www.kernel.org/doc/html/latest/admin-guide/laptops/thinkpad-acpi.html#id7):

> Readings from sensors that are not available return -128.

There should be a way to ignore this measurement.

vmatare, 2021-02-04:
Well, the number in brackets behind the actual temperature is a bias that has been added due to the sudden "jump" in temperatures caused by the -128 artifact. You can disable biasing by running thinkfan with the `-b0` commandline option.

rnhmjoj, 2021-02-07:
Thank you, that works. I would still leave this open, though: I think the value should be ignored, at least if the sensors is `tpacpi`.

vmatare, 2021-04-16:
@ratcashdev should now also be affected by this issue. Could either @rnhmjoj or @ratcashdev (or preferrably both) give the updated master branch another go and let me know whether this issue is fixed for you? I'll then make a bugfix release so that the fixes for #111 and #123 get picked up by the distributions.

ratcashdev, 2021-04-16:
does not seem to help:
```
sudo ./thinkfan -n -v
/sys/devices/platform/thinkpad_hwmon/hwmon/hwmon6/pwm1: Saved initial state: 2.
Temperatures(bias): 42(0) -> Fans: 32
Going to sleep: Will allow sensor read errors for the next 4 loops.
Lost sensor read_temps: Failed to read temperature(s) from /sys/devices/platform/thinkpad_hwmon/hwmon/hwmon6/temp1_input: No such device or address
Temperatures(bias): -128(0) -> Fans: 0
Received SIGUSR2: Re-initializing fan control.
Lost sensor read_temps: Failed to read temperature(s) from /sys/devices/platform/thinkpad_hwmon/hwmon/hwmon6/temp1_input: No such device or address
Temperatures(bias): -128(0) -> Fans: 0
^CCleaning up and resetting fan control.
/sys/devices/platform/thinkpad_hwmon/hwmon/hwmon6/pwm1: Restoring initial state: 2.
```

rnhmjoj, 2022-04-13:
I just checked by removing the workaround you previously suggested (`-b 0`) and the issue seems resolved.
Thank you again.