vmatare/thinkfan #126: Refusing to run without usable config file!

kendoori, 2021-02-09:
Am running TF on Linux Mint 20.1

Running: service thinkfan status yields the following

[image]

My config file was built from this list
[image]

My thinkfan.conf looks like this:
[image]

kendoori, 2021-02-10:
FYI, the config here worked: https://gist.github.com/Yatoom/1c80b8afe7fa47a938d3b667ce234559#gistcomment-3306472

vmatare, 2021-04-15:
This looks a lot like thinkfan might be trying to start before the necessary modules have been loaded. That said, the 0.9.1 release is ancient, so please try with a newer version (and tell your distribution to update the thinkfan package). If the problem persists with the latest thinkfan release, you're probably experiencing issue #118. So then you'd want to read up on #118 and try the workarounds mentioned there until I've found the time to implement a proper fix.

closed 2021-04-15 by vmatare