vmatare/thinkfan #127: Thinkfan on Thinkpad E14 Gen 2

limmor1, 2021-02-12:
Hi! Thinkfan seems to be a great program, but I am having one issue - even when I set the fan to work at a single speed for all temperatures I am getting periodic drops in RPM. The fan is running at the specified level, but then periodically switches to "auto" and the RPMs drop for a short moment and then resume again.

Things I tried:
- changing `thinkfan -s 5` to different periodicity. The periodicity depends on the time, e.g. if I set "1", the RPMs drop more often. If I set "15" the drops are less frequent.
- polled` /proc/acpi/ibm/thermal` to verify that my temperature sensor does not register any drops or spikes (I polled with bash loop, so there there is a limit to how frequently I can poll, but I did not see any sudden changes in temperature)
- polled` /proc/acpi/ibm/fan` to verify that the fan reports the same level all the time, except during the drop it changes to "auto".

My setup:
- Thinkpad E14 Gen 2.
- Linux Mint, kernel 5.8.0-40 to 43.
- thinkfan 0.9.3.2

/etc/thinkfan.conf:
```
tp_fan /proc/acpi/ibm/fan
tp_thermal /proc/acpi/ibm/thermal (10 0 0 0 0 0 0 0)
(5, -32767, 32767)  # (also tried (127, 0, 256) in case that matters, but same effect

```
/etc/default/thinkfan:
`DAEMON_ARGS="-q -b 5.0 -s 5"`

vmatare, 2021-02-12:
This sounds a lot like this issue: https://github.com/vmatare/thinkfan/issues/114#issuecomment-757535018

That would mean the problem lies with either the `thinkpad_acpi` kernel module or with the laptop's embedded controller firmware. A possible workaround might be to have thinkfan disable the watchdog. Currently that's not allowed because it's potentially dangerous if thinkfan or your entire userspace freezes.

limmor1, 2021-02-13:
Thanks! This made me aware that apt repository has a very old version of thinkfan. After building new version I no longer experience the issue.

closed 2021-02-13 by limmor1