vmatare/thinkfan #128: ERROR: Unhandled std::length_error: basic_string::_M_replace_aux.

marco-parillo, 2021-02-25:
```
$ sudo thinkfan -n
[sudo] password for mparillo: 
ERROR: Unhandled std::length_error: basic_string::_M_replace_aux.
errno = No such file or directory.
Backtrace:
thinkfan(+0x37835) [0x556f9b74e835]
thinkfan(+0x205b2) [0x556f9b7375b2]
/usr/lib/libstdc++.so.6(+0xa2d3a) [0x7f3a7a74dd3a]
/usr/lib/libstdc++.so.6(+0xa2da7) [0x7f3a7a74dda7]
/usr/lib/libstdc++.so.6(+0xa304e) [0x7f3a7a74e04e]
/usr/lib/libstdc++.so.6(std::__throw_length_error(char const*)+0x45) [0x7f3a7a74455f]
/usr/lib/libstdc++.so.6(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::append(unsigned long, char)+0) [0x7f3a7a7e9ca0]
thinkfan(thinkfan::Config::try_read_config(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0x202) [0x556f9b73e4d2]
thinkfan(thinkfan::Config::read_config(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&)+0x1a) [0x556f9b73f4da]
thinkfan(main+0x311) [0x556f9b738731]
/usr/lib/libc.so.6(__libc_start_main+0xd5) [0x7f3a7a4e9b25]
thinkfan(_start+0x2e) [0x556f9b73916e]

This is probably a bug. Please consider reporting this at https://github.com/vmatare/thinkfan/issues. Thanks.
Aborted
```
If it helps:

```
$ inxi -Fazy
System:
  Kernel: 5.11.1-arch1-1 x86_64 bits: 64 compiler: gcc v: 10.2.1 
  parameters: BOOT_IMAGE=/vmlinuz-linux 
  root=UUID=5254b91f-12c0-431c-bb4b-92f786328283 rw loglevel=3 
  Desktop: KDE Plasma 5.21.1 tk: Qt 5.15.2 wm: kwin_x11 dm: SDDM 
  Distro: Arch Linux 
Machine:
  Type: Laptop System: LENOVO product: 2522BH1 v: ThinkPad T410 
  serial: <filter> Chassis: type: 10 serial: <filter> 
  Mobo: LENOVO model: 2522BH1 serial: <filter> BIOS: LENOVO 
  v: 6IET84WW (1.44 ) date: 09/26/2012 
Battery:
  ID-1: BAT0 charge: 15.2 Wh condition: 15.2/93.2 Wh (16%) volts: 12.6/11.1 
  model: SANYO 42T4799 type: Li-ion serial: <filter> status: Full 
CPU:
  Info: Dual Core model: Intel Core i5 M 520 bits: 64 type: MT MCP 
  arch: Nehalem family: 6 model-id: 25 (37) stepping: 5 microcode: 3 
  L2 cache: 3 MiB 
  flags: lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 bogomips: 19156 
  Speed: 1436 MHz min/max: 1199/2400 MHz boost: enabled Core speeds (MHz): 
  1: 1436 2: 1403 3: 1246 4: 1463 
  Vulnerabilities: Type: itlb_multihit status: KVM: VMX unsupported 
  Type: l1tf mitigation: PTE Inversion 
  Type: mds 
  status: Vulnerable: Clear CPU buffers attempted, no microcode; SMT vulnerable 
  Type: meltdown mitigation: PTI 
  Type: spec_store_bypass status: Vulnerable 
  Type: spectre_v1 
  mitigation: usercopy/swapgs barriers and __user pointer sanitization 
  Type: spectre_v2 
  mitigation: Full generic retpoline, STIBP: disabled, RSB filling 
  Type: srbds status: Not affected 
  Type: tsx_async_abort status: Not affected 
Graphics:
  Device-1: Intel Core Processor Integrated Graphics vendor: Lenovo 
  driver: i915 v: kernel bus ID: 00:02.0 chip ID: 8086:0046 class ID: 0300 
  Device-2: Chicony Integrated Camera type: USB driver: uvcvideo 
  bus ID: 1-1.6:5 chip ID: 17ef:480f class ID: 0e02 
  Display: x11 server: X.Org 1.20.10 compositor: kwin_x11 driver: 
  loaded: intel unloaded: modesetting alternate: fbdev,vesa display ID: :0 
  screens: 1 
  Screen-1: 0 s-res: 1440x900 s-dpi: 96 s-size: 381x238mm (15.0x9.4") 
  s-diag: 449mm (17.7") 
  Monitor-1: LVDS1 res: 1440x900 hz: 60 dpi: 122 size: 300x190mm (11.8x7.5") 
  diag: 355mm (14") 
  Message: Unable to show advanced data. Required tool glxinfo missing. 
Audio:
  Device-1: Intel 5 Series/3400 Series High Definition Audio vendor: Lenovo 
  driver: snd_hda_intel v: kernel bus ID: 00:1b.0 chip ID: 8086:3b56 
  class ID: 0403 
  Sound Server: ALSA v: k5.11.1-arch1-1 
Network:
  Device-1: Intel 82577LM Gigabit Network vendor: Lenovo driver: e1000e 
  v: kernel port: 1820 bus ID: 00:19.0 chip ID: 8086:10ea class ID: 0200 
  IF: enp0s25 state: down mac: <filter> 
  Device-2: Intel Centrino Advanced-N 6200 driver: iwlwifi v: kernel 
  port: 1880 bus ID: 03:00.0 chip ID: 8086:4239 class ID: 0280 
  IF: wlp3s0 state: up mac: <filter> 
Bluetooth:
  Device-1: Broadcom Corp Broadcom Bluetooth Device type: USB driver: btusb 
  v: 0.8 bus ID: 1-1.4:4 chip ID: 0a5c:217f class ID: fe01 serial: <filter> 
  Message: Required tool hciconfig not installed. Check --recommends 
Drives:
  Local Storage: total: 149.05 GiB used: 28.34 GiB (19.0%) 
  SMART Message: Unable to run smartctl. Root privileges required. 
  ID-1: /dev/sda maj-min: 8:0 vendor: Hitachi model: HTS725016A9A364 
  size: 149.05 GiB block size: physical: 512 B logical: 512 B speed: 3.0 Gb/s 
  rotation: 7200 rpm serial: <filter> rev: C70F scheme: MBR 
Partition:
  ID-1: / raw size: 148.34 GiB size: 145.01 GiB (97.76%) 
  used: 28.27 GiB (19.5%) fs: ext4 dev: /dev/sda2 maj-min: 8:2 
  ID-2: /boot raw size: 212 MiB size: 201.3 MiB (94.95%) 
  used: 72.8 MiB (36.2%) fs: ext4 dev: /dev/sda1 maj-min: 8:1 
Swap:
  Kernel: swappiness: 60 (default) cache pressure: 100 (default) 
  ID-1: swap-1 type: partition size: 512 MiB used: 0 KiB (0.0%) priority: -2 
  dev: /dev/sda3 maj-min: 8:3 
Sensors:
  System Temperatures: cpu: 55.0 C mobo: 0.0 C 
  Fan Speeds (RPM): cpu: 3685 
Info:
  Processes: 172 Uptime: 5h 24m wakeups: 1 Memory: 3.64 GiB 
  used: 1.58 GiB (43.3%) Init: systemd v: 247 Compilers: gcc: 10.2.0 
  clang: 11.1.0 Packages: pacman: 892 lib: 194 Shell: Bash v: 5.1.4 
  running in: konsole inxi: 3.3.01 
```

vmatare, 2021-04-15:
Hi @marco-parillo, thanks for reporting. This issue looks unfamiliar to me and the auto-generated stack trace doesn't give me enough of a hint at what could be going wrong. Could you please:

1. Post your thinkfan config file
2. After thinkfan has crashed, generate a better stacktrace with: `coredumpctl debug thinkfan -A "-ex bt -ex quit"`

vmatare, 2022-04-09:
Closing due to lack of information that would allow reproducing the issue.

TommiRommi, 2023-10-05:
funny how i have this exact same issue
thinkfan config file
```
##############################################################################
# thinkfan Example Config File
# ============================
#
# Please read the config manpage thinkfan.conf(5) before playing around with
# this.
#
# This is NOT a working config file that can just be copied. It is only meant
# to give a rough idea what can be done. In particular, don't copy & paste the
# fan speed config! Think about what you're doing.
#
# If you don't know what temperatures are right for your system, you should
# not be using thinkfan!
##############################################################################

##############################################################################
# Sensor Drivers and Temperature Inputs
# =====================================
#
# ATTENTION: The order in which sensors are specified here is significant when
# specifying the fan speeds further below!
#
# There are multiple ways in which a temperature input can be specified. An
# example for each is given below.
#
# The "correction:" and "optional:" keywords may be specified on any type of
# sensor.

sensors:
  # LM Sensors
  # ==========
  # Temperatures can be read directly from Linux drivers through the LM sensors.
  #
  # To configure this, install "lm-sensors" and "libsensors", then
  # run "sensors-detect", then run "sensors".
  # To build thinkfan from sources, you'll also need to install "libsensors-dev"
  # or equivalent package for your distribution.
  #
  # For example, the following output of "sensors":
  # ...
  # thinkpad-isa-0000
  # Adapter: ISA adapter
  # fan1:        2618 RPM
  # fan2:        2553 RPM
  # CPU:          +63.0 C
  # GPU 1:        +55.0 C
  # temp3:        +68.0 C
  # temp4:         +0.0 C
  # temp5:        +60.0 C
  # temp6:        +64.0 C
  # temp7:        +67.0 C
  # temp8:         +0.0 C
  # ...
  # would result in the following configuration:
  - chip: thinkpad-isa-0000
    ids: [ CPU, "GPU 1", temp3, temp4, temp5, temp6, temp7, temp8 ]

  # hwmon: Full path to a temperature file (single sensor).
  # =======================================================
  # Disadvantage is that the index in "hwmon0" depends on the load order of
  # the driver modules, which may change across bootups on some systems.
  - hwmon: /sys/class/hwmon/hwmon0/temp1_input

  # hwmon: Path to a complete driver folder
  # =======================================
  # Individual sensors need to be picked out with the "indices:" keyword.
  # This can be used with a stable path that does not depend on driver load
[... 250 lines cut ...]
    scheme: GPT
  ID-2: /dev/sda maj-min: 8:0 vendor: Kingston model: DataTraveler 3.0
    size: 28.88 GiB block-size: physical: 512 B logical: 512 B type: USB rev: 3.2
    spd: 5 Gb/s lanes: 1 mode: 3.2 gen-1x1 tech: N/A serial: <filter>
    fw-rev: PMAP scheme: MBR
Partition:
  ID-1: / raw-size: 221.97 GiB size: 217.43 GiB (97.95%) used: 6.14 GiB (2.8%)
    fs: ext4 dev: /dev/nvme0n1p3 maj-min: 259:3
  ID-2: /boot raw-size: 512 MiB size: 511 MiB (99.80%)
    used: 111.6 MiB (21.8%) fs: vfat dev: /dev/nvme0n1p1 maj-min: 259:1
Swap:
  Kernel: swappiness: 60 (default) cache-pressure: 100 (default) zswap: yes
    compressor: zstd max-pool: 20%
  ID-1: swap-1 type: partition size: 16 GiB used: 0 KiB (0.0%) priority: -2
    dev: /dev/nvme0n1p2 maj-min: 259:2
Sensors:
  System Temperatures: cpu: 44.0 C pch: 49.0 C mobo: N/A
  Fan Speeds (rpm): fan-1: 0
Info:
  Processes: 197 Uptime: 57m wakeups: 3 Memory: total: 16 GiB note: est.
  available: 15.26 GiB used: 1.85 GiB (12.1%) Init: systemd v: 254
  default: graphical tool: systemctl Compilers: gcc: 13.2.1 Packages:
  pm: pacman pkgs: 481 libs: 143 tools: yay Shell: Zsh v: 5.9
  running-in: alacritty inxi: 3.3.30
```