vmatare/thinkfan #136: What the correct device id is, how to find it out?

naddika, 2021-06-23:
Arch Linux, Think Pad yoga 11e

It won't run because the fan device is incorrect.

```
sudo systemctl status thinkfan

 thinkfan.service - simple and lightweight fan control program
     Loaded: loaded (/usr/lib/systemd/system/thinkfan.service; disabled; vendor preset: disabled)
    Drop-In: /etc/systemd/system/thinkfan.service.d
             └─override.conf
     Active: failed (Result: exit-code) since Wed 2021-06-23 21:40:27 +03; 1s ago
    Process: 52494 ExecStart=/usr/bin/thinkfan $THINKFAN_ARGS (code=exited, status=1/FAILURE)
        CPU: 5ms

Jun 23 21:40:27 user1 systemd[1]: Starting simple and lightweight fan control program...
Jun 23 21:40:27 user1 thinkfan[52494]: ERROR: Error scanning /sys/devices/pci0000:00/0000:00:03.1/0000:27:00.0/hwmon: No such file or directory
Jun 23 21:40:27 user1 systemd[1]: thinkfan.service: Control process exited, code=exited, status=1/FAILURE
Jun 23 21:40:27 user1 systemd[1]: thinkfan.service: Failed with result 'exit-code'.
Jun 23 21:40:27 user1 systemd[1]: Failed to start simple and lightweight fan control program.

```

How do I find out what the correct device is?

---

cat /usr/share/doc/thinkfan/examples/thinkfan.yaml

```
sensors:
  - hwmon: /sys/class/hwmon/hwmon0/temp1_input

  - hwmon: /sys/devices/pci0000:00/0000:00:03.1/0000:27:00.0/hwmon
    indices: [1, 2, 5, 6] # adds 4 temperature sensors
    correction: [0, 0, 0, -5] # add -5 °C to temp6_input

  - hwmon: /sys/class/hwmon
    name: k10temp
    indices: [1]
  # [..............]

```

vmatare, 2021-06-23:
Try running `sensors` as root to get a first overview and then go ahead and explore your `/sys/class/hwmon` directory. In case you've never looked at sysfs before: You basically just wanna output the contents of the various files in there, that'll give you an idea what's going on. Basically you're looking for files called `temp*_input`. Take a look at the manpage (`man thinkfan.conf`) and PLEASE DON'T just copy an example config. Figuring out why it doesn't work for you (it never will) is probably harder than reading the manpage and writing a correct one from scratch.

vmatare, 2021-06-23:
Also very important documentation on `/sys/class/hwmon`: https://www.kernel.org/doc/html/latest/hwmon/sysfs-interface.html

naddika, 2021-06-24:
I've run sensors. And this:

```
 tree /sys/class/hwmon
/sys/class/hwmon
├── hwmon0 -> ../../devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0003:00/power_supply/ACAD/hwmon0
├── hwmon1 -> ../../devices/virtual/thermal/thermal_zone0/hwmon1
├── hwmon2 -> ../../devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT1/hwmon2
├── hwmon3 -> ../../devices/platform/thinkpad_hwmon/hwmon/hwmon3
├── hwmon4 -> ../../devices/virtual/thermal/thermal_zone1/hwmon4
├── hwmon5 -> ../../devices/platform/coretemp.0/hwmon/hwmon5
├── hwmon6 -> ../../devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-WCOM5071:00/0018:056A:5074.0001/power_supply/wacom_battery_0/hwmon6
└── hwmon7 -> ../../devices/virtual/thermal/thermal_zone3/hwmon7

8 directories, 0 files
```

There's no  `temp*_input`

vmatare, 2021-06-24:
yeah, these are probably symlinks and it looks like your `tree` program isn't following them.

vmatare, 2021-06-24:
take a look at the help with `tree -h`. You'll want to look at the `-l` option to make it follow symlinks and the `-L` option to limit the maximum depth. Try `tree -l -L 2` for example.

naddika, 2021-06-28:
```
% tree -l -L 2 /sys/class/hwmon
/sys/class/hwmon
├── hwmon0 -> ../../devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0003:00/power_supply/ACAD/hwmon0
│   ├── device -> ../../ACAD
│   ├── name
│   ├── power
│   ├── subsystem -> ../../../../../../../class/hwmon  [recursive, not followed]
│   └── uevent
├── hwmon1 -> ../../devices/virtual/thermal/thermal_zone0/hwmon1
│   ├── device -> ../../thermal_zone0
│   ├── name
│   ├── power
│   ├── subsystem -> ../../../../../class/hwmon  [recursive, not followed]
│   ├── temp1_crit
│   ├── temp1_input
│   └── uevent
├── hwmon2 -> ../../devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT1/hwmon2
│   ├── device -> ../../BAT1
│   ├── in0_input
│   ├── name
│   ├── power
│   ├── subsystem -> ../../../../../../../class/hwmon  [recursive, not followed]
│   └── uevent
├── hwmon3 -> ../../devices/virtual/thermal/thermal_zone1/hwmon3
│   ├── device -> ../../thermal_zone1
│   ├── name
│   ├── power
│   ├── subsystem -> ../../../../../class/hwmon  [recursive, not followed]
│   ├── temp1_input
│   └── uevent
├── hwmon4 -> ../../devices/platform/thinkpad_hwmon/hwmon/hwmon4
│   ├── device -> ../../../thinkpad_hwmon
│   ├── fan1_input
│   ├── name
│   ├── power
│   ├── pwm1
│   ├── pwm1_enable
│   ├── subsystem -> ../../../../../class/hwmon  [recursive, not followed]
│   ├── temp10_input
│   ├── temp11_input
│   ├── temp12_input
│   ├── temp13_input
│   ├── temp14_input
│   ├── temp15_input
│   ├── temp16_input
│   ├── temp1_input
│   ├── temp2_input
│   ├── temp3_input
│   ├── temp4_input
│   ├── temp5_input
│   ├── temp6_input
│   ├── temp7_input
│   ├── temp8_input
│   ├── temp9_input
│   └── uevent
├── hwmon5 -> ../../devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-WCOM5071:00/0018:056A:5074.0001/power_supply/wacom_battery_0/hwmon5
│   ├── device -> ../../wacom_battery_0
│   ├── name
│   ├── power
│   ├── subsystem -> ../../../../../../../../../../class/hwmon  [recursive, not followed]
│   └── uevent
├── hwmon6 -> ../../devices/virtual/thermal/thermal_zone2/hwmon6
│   ├── device -> ../../thermal_zone2
│   ├── name
│   ├── power
│   ├── subsystem -> ../../../../../class/hwmon  [recursive, not followed]
│   ├── temp1_input
│   └── uevent
└── hwmon7 -> ../../devices/platform/coretemp.0/hwmon/hwmon7
    ├── device -> ../../../coretemp.0
    ├── name
    ├── power
    ├── subsystem -> ../../../../../class/hwmon  [recursive, not followed]
    ├── temp1_crit
    ├── temp1_crit_alarm
    ├── temp1_input
    ├── temp1_label
    ├── temp1_max
    ├── temp2_crit
    ├── temp2_crit_alarm
    ├── temp2_input
    ├── temp2_label
    ├── temp2_max
    ├── temp3_crit
    ├── temp3_crit_alarm
    ├── temp3_input
    ├── temp3_label
    ├── temp3_max
    └── uevent

```

naddika, 2021-06-28:
there're multiple `temp*_input`

closed 2021-06-28 by vmatare