vmatare/thinkfan #137: Segmentation fault

josgutie, 2021-06-28:
ERROR: Segmentation fault.
Backtrace:
thinkfan(+0x29a63) [0x55f99bb3fa63]
thinkfan(thinkfan::Error::Error(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0x51) [0x55f99bb3ff61]
thinkfan(thinkfan::Bug::Bug(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0xd) [0x55f99bb400cd]
thinkfan(+0x123b7) [0x55f99bb283b7]
/lib64/libpthread.so.0(+0x13a20) [0x7f3fec27ba20]
thinkfan(thinkfan::ComplexLevel::down() const+0x2d) [0x55f99bb3362d]
thinkfan(thinkfan::StepwiseMapping::init_fanspeed(thinkfan::TemperatureState const&)+0x3e) [0x55f99bb3480e]
thinkfan(thinkfan::run(thinkfan::Config const&)+0x78) [0x55f99bb396d8]
thinkfan(main+0x309) [0x55f99bb32879]
/lib64/libc.so.6(__libc_start_main+0xd5) [0x7f3febe1db75]
thinkfan(_start+0x2e) [0x55f99bb334ce]

I've attached the configuration file, it's a Lenovo P1 Gen3.

vmatare, 2021-06-28:
@josgutie I think your attachment got lost somehow. Could you just copy&paste your config in a comment, please?

josgutie, 2021-06-28:
[thinkfan.conf.gz](https://github.com/vmatare/thinkfan/files/6728521/thinkfan.conf.gz)

josgutie, 2021-06-28:
Config file uploaded.

Thank you very much

boytunp, 2021-08-23:
```
\ERROR: Segmentation fault.
Backtrace:
thinkfan(+0x3c9e5) [0x55bf910a49e5]
thinkfan(thinkfan::Bug::Bug(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0x51) [0x55bf910a6be1]
thinkfan(+0x1c05d) [0x55bf9108405d]
/usr/lib/libpthread.so.0(+0x13870) [0x7f9a1436f870]
thinkfan(thinkfan::ComplexLevel::down() const+0x30) [0x55bf91090a30]
thinkfan(thinkfan::StepwiseMapping::init_fanspeed(thinkfan::TemperatureState const&)+0x3e) [0x55bf91090cce]
thinkfan(thinkfan::run(thinkfan::Config const&)+0x80) [0x55bf9108d670]
thinkfan(main+0x363) [0x55bf9108b9c3]
/usr/lib/libc.so.6(__libc_start_main+0xd5) [0x7f9a13f09b25]
thinkfan(_start+0x2e) [0x55bf9108c35e]

This is probably a bug. Please consider reporting this at https://github.com/vmatare/thinkfan/issues. Thanks.

```
```

sensors:
  - hwmon: /sys/class/hwmon
    name: coretemp 
    indices: [1]
################################################################
fans:
  - tpacpi: /proc/acpi/ibm/fan

levels:
  - speed: 0
    upper_limit: 47

  - speed: 1
    lower_limit: 47
    upper_limit: 55

  - speed: 2
    lower_limit: 55
    upper_limit: 60

  - speed: 4
    lower_limit: 60
    upper_limit: 65

  - speed: 5
    lower_limit: 65
    upper_limit: 73

  - speed: 6
    lower_limit: 73
    upper_limit: 80

  - speed: 7
    lower_limit: 80

```

vmatare, 2021-08-29:
There is an error in your config which wasn't being handled properly. I've pushed some updates that should give you a helpful error message. @josgutie, @boytunp, please test and let me know whether the message is helpful.

boytunp, 2021-09-09:
Yes now it correctly shows the errors in the config. TYVM!

closed 2021-09-09 by vmatare