vmatare/thinkfan #139: thinkfan terminated by SIGABRT

karthink, 2021-06-29:
```sh
$ sudo thinkfan -n
/usr/include/c++/11.1.0/bits/stl_vector.h:1063: std::vector<_Tp, _Alloc>::const_reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) const [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::const_reference = const int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
fish: Job 1, 'sudo thinkfan -n' terminated by signal SIGABRT (Abort)
```

```bash
$ cat /etc/modprobe.d/thinkfan.conf
options thinkpad_acpi fan_control=1
```

Installed from the Arch AUR (builds from source). 

Config file (`/etc/thinkfan.yaml`)

```yaml
sensors:
  - tpacpi: /proc/acpi/ibm/thermal
    indices: [1]

fans:
  - tpacpi: /proc/acpi/ibm/fan

levels:
  - [0, 0, 58]
  - ["level auto", 56, 80]
  - ["level disengaged", 78, 255]
````

Thinkpad: T14 Gen 1 AMD

haaase, 2021-07-05:
I can confirm this issue. I'm experiencing the same error on the following setup: Thinkpad T14 Gen 2 Intel, Arch Linux (5.12.13 kernel), thinkfan v1.2.2.

Interestingly, I could fix the issue by removing the `indices` line. It also works when using a `hwmon` based config instead of `tpacpi`, i.e., both of the configuration options below work on my machine:

```YAML
sensors:
  - hwmon: /sys/class/hwmon
    name: coretemp
    indices: [1,2,3,4,5]

  - tpacpi: /proc/acpi/ibm/thermal
    # indices: [1,2,3]
```

karthink, 2021-07-07:
That's useful to know. I'll try this and report if it works for me too.

vmatare, 2021-08-29:
This should be fixed in the latest master, specifically by https://github.com/vmatare/thinkfan/commit/6b9e34895064b7aede360958649910de4839ab8a. Please test and let me know if the issue is fixed for you.