vmatare/thinkfan #149: Manjaro Linux couldn't get thinkfan working

viper2035, 2021-08-20:
Hello everyone,
I just started with using Manjaro Linux on my T430S. I used different other Linux distributions before mostly debian based. Now I tried to get thinkfan working on Manjaro Linux. I followed this [guide](https://wiki.archlinux.org/title/fan_speed_control#ThinkPad_laptops). But at the point of using the command `thinkfan -n` i got the message:
```
/usr/include/c++/11.1.0/bits/stl_vector.h:1063: std::vector<_Tp, _Alloc>::const_reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) const [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::const_reference = const int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Abgebrochen (Speicherabzug geschrieben)
```
If I enable the service of thinkfan I got the error message after using `systemctl status thinkfan.service`:
```
× thinkfan.service - simple and lightweight fan control program
     Loaded: loaded (/usr/lib/systemd/system/thinkfan.service; enabled; vendor preset: disabled)
    Drop-In: /etc/systemd/system/thinkfan.service.d
             └─override.conf
     Active: failed (Result: core-dump) since Fri 2021-08-20 08:25:49 CEST; 2h 53min ago
    Process: 389 ExecStart=/usr/bin/thinkfan $THINKFAN_ARGS (code=dumped, signal=ABRT)
        CPU: 6ms

Aug 20 08:25:48 Lenovot430slinux systemd[1]: Starting simple and lightweight fan control program...
Aug 20 08:25:49 Lenovot430slinux systemd[1]: thinkfan.service: Control process exited, code=dumped, status=6/ABRT
Aug 20 08:25:49 Lenovot430slinux systemd[1]: thinkfan.service: Failed with result 'core-dump'.
Aug 20 08:25:49 Lenovot430slinux systemd[1]: Failed to start simple and lightweight fan control program
```
But I have no glue whats the problem here. My thinkfan.conf isn't that long:
sensors:
```
  - tpacpi: /proc/acpi/ibm/thermal
    indices: [1]

fans:
  - tpacpi: /proc/acpi/ibm/fan
levels:
  - ["level 0", 0, 70]
  - ["level 1", 65, 73]
  - ["level 3", 70, 80]
  - ["level 5", 75, 88]
  - ["level auto", 83, 255]
```
Using the commands to read the temperature and fan speed works fines (`cat /proc/acpi/ibm/thermal` `cat /proc/acpi/ibm/fan`). And I can set the fan speed by using `echo level 1 > /proc/acpi/ibm/fan`.
Does anybody got an idea what I did wrong?

Thank you all

vmatare, 2021-08-29:
This seems to be the same problem as in #139, which should be fixed in the current master branch. It'd be very helpful if you could test with that and report back whether it helps.

vmatare, 2022-03-29:
Closing for lack of feedback and since the symptoms match fixed & released #139.

closed 2022-03-29 by vmatare