vmatare/thinkfan #155: ERROR: No fans are configured in /etc/thinkfan.conf

cgirerd, 2021-09-22:
Hello,
Thank you very much for this software! I just downloaded and compiled the code of the master branch. I the installed it on my ThinkPad P53 running Debian 11. I ran thinkfan using the command `thinkfan -n` and get the following error:
`ERROR: No fans are configured in /etc/thinkfan.conf`
However, the fans seem to be configured correctly in /etc/thinkfan.conf to me. Here is the content of this file:

```
sensors:

  - hwmon: /sys/devices/platform/thinkpad_hwmon/hwmon
    name: thinkpad
    indices: [1, 2, 3, 4, 5, 6, 7, 8]

  - hwmon: /sys/devices/platform/coretemp.0/hwmon
    name: coretemp
    indices: [1, 2, 3, 4, 5, 6, 7]

  - hwmon: /sys/devices/pci0000:00/0000:00:1b.0/0000:02:00.0/hwmon
    name: nvme
    indices: [1, 2]

  - hwmon: /sys/devices/pci0000:00/0000:00:1d.0/0000:55:00.0/hwmon
    name: nvme
    indices: [1, 2]

  - hwmon: /sys/devices/virtual/thermal/thermal_zone0/hwmon0
    name: acpitz
    indices: [1]

  - hwmon: /sys/devices/virtual/thermal/thermal_zone9
    name: iwlwifi_1
    optional: true
    indices: [1]

  - hwmon: /sys/devices/virtual/thermal/thermal_zone6
    name: pch_cannonlake
    indices: [1]

  - nvml: 01:00.0

fans:
  - tpacpi: /proc/acpi/ibm/fan

levels:
  - [0, 0, 55]
  - [1, 50, 60]
  - [2, 57, 63]
  - [3, 60, 66]
  - [4, 63, 69]
  - [5, 66, 72]
  - [6, 69, 75]
  - [7, 72, 32767]
```
I would be happy to have any hint on this issue... :)

koutheir, 2021-09-22:
Can you provide the output of this command?
```shell
$ cat /proc/acpi/ibm/fan
```

cgirerd, 2021-09-22:
Thanks for the fast reply! Here is the output of `$ cat /proc/acpi/ibm/fan`:
```
status:		enabled
speed:		2037
level:		auto
commands:	level <level> (<level> is 0-7, auto, disengaged, full-speed)
commands:	enable, disable
commands:	watchdog <timeout> (<timeout> is 0 (off), 1-120 (seconds))
```
The ThinkPad P53 is also a dual fan system in case it has some importance at some point.

koutheir, 2021-09-22:
Can you provide the output of this command?
```sh
$ /usr/local/sbin/thinkfan -v -n -c /etc/thinkfan.conf
```

vmatare, 2021-09-22:
Thanks for your support @koutheir, but this was simply an oversight on my part. The config syntax (and its semantics) has gotten quite complex with the multifan support and the design of the YAML-based config parser hasn't really evolved with that. It's becoming very apparent that I need to put some end-to-end tests in the CI workflow.

Anyways, I've pushed a fix to the master branch. Sorry for the inconvenience and thanks for reporting! @cgirerd Let me know what results you're getting.

cgirerd, 2021-09-23:
Thanks for pushing the fix, the issue is gone, all works well!

Actually it worked well when I ran the thinkfan manually using `thinkfan -n`, but when I enabled thinkfan in systemd it had some issues finding some sensors like the one on my wireless card. This particular sensor was actually marked as optional as in my initial post, but thinkfan couldn't find the file "name" that contains "iwlwifi_1". I guess that thinkfan was starting before some of the needed services like the networking service in the case of my wireless card. I fixed the issue temporarily by adding a delay in "/etc/systemd/system/thinkfan.service.d/default.conf" under [service], and will try to do something cleaner by finding out the specific services that need to be started before thinkfan. [service] now looks like that at the moment:
```
[Service]
# Decrease biasing (up to -b-10) if your fan speed changes too quickly,
# Increase biasing (up to -b20) if your fan speed changes too slowly.
Environment='THINKFAN_ARGS=-b0'
ExecStartPre=/bin/sleep 30
```
Thanks again @vmatare and @koutheir for your help and fast replies!

vmatare, 2021-09-23:
> Actually it worked well when I ran the thinkfan manually using `thinkfan -n`, but when I enabled thinkfan in systemd it had some issues finding some sensors like the one on my wireless card. This [...]

Yeah, that is a long-standing issue that many people are having, but it's not actually thinkfan's responsibility to check whether all its dependent services are ready. However, by now I'm close to actually implementing a workaround since the problem isn't trivial to resolve in systemd (where it belongs) either.

Anyways, I'm closing this since the original issue has been fixed. See e.g. issue #118 if you want to join the discussion on thinkfan being started too early.

closed 2021-09-23 by vmatare