vmatare/thinkfan #156: Found multiple hwmons with this name "nvme"

koutheir, 2021-10-05:
I configured my `thinkfan.yaml` to take into account the NVME temperatures like follows:
```yaml
sensors:
  - hwmon: /sys/class/hwmon
    name: nvme
    indices: [2, 3]
    correction: [0, 0]
```

But running `thinkfan -nv` issues this error:
```
ERROR: /etc/thinkfan.conf:37:
    name: nvme
          ^
Found multiple hwmons with this name:  /sys/class/hwmon/hwmon2 /sys/class/hwmon/hwmon1.
```

And it is true that both `/sys/class/hwmon/hwmon1/name` and `/sys/class/hwmon/hwmon2/name` have the same contents, even though they correspond to the two distinct NVME slots on the mother board.

How can I solve this issue?

koutheir, 2021-10-06:
> How can I solve this issue?

I know my question is vague, but it's because I lack information on how this could be solved correctly. I'm ready to create a pull request if we can agree on some approach to fix it.

Should we, for example, filter `hwmon`s by something more than names? Or may be accept multiple `hwmon`s having the same name and handle this situation as *expected* instead of an error? Or maybe perform special behavior for some file system elements?

koutheir, 2021-10-12:
I think a reliable solution to this issue (and I suspect other issues too), would be to use `libsensors` as a source of temperatures. Using `libsensors`, my two NVME slots are identified as `nvme-pci-0200` and `nvme-pci-5500`.

vmatare, 2021-10-15:
Yeah, I suspected this would come up eventually. I'd be super happy if you were to submit a pull request implementing sensor discovery via `libsensors`. My only condition is that `libsensors` should be an optional build-time dependency ;-)

vmatare, 2022-08-08:
Implemented in master and slated for 2.0

closed 2022-08-08 by vmatare