vmatare/thinkfan #165: thinkfan does not start (thinkpad yoga x1 gen6)

salvoc81, 2021-11-17:
I am trying to use thinkfan on a x1 yoga gen 6 but I get the following error:  

```
sudo thinkfan -n

ERROR: ~TpFanDriver: Resetting fan control in /proc/acpi/ibm/fan: Invalid argument
ERROR: Module thinkpad_acpi doesn't seem to support fan_control
```

Any idea in what causing the issue?
Thanks a lot.

vmatare, 2021-11-17:
https://www.kernel.org/doc/html/latest/admin-guide/laptops/thinkpad-acpi.html?highlight=thinkpad_acpi#fan-control-and-monitoring-fan-speed-fan-enable-disable

vmatare, 2021-11-17:
maybe I should put that link right in the error message...

vmatare, 2022-11-22:
The error message is now significantly extended, hopefully making it clearer what the user needs to do in this situation...

closed 2022-11-22 by vmatare