vmatare/thinkfan #166: permission denied when trying to set the level (x1 yoga gen 6)

salvoc81, 2021-11-17:
When I try to set level I get the following error:

```
sudo echo level 1 > /proc/acpi/ibm/fan
bash: /proc/acpi/ibm/fan: Permission denied
```
Any suggestion regarding possible solutions?
Thanks a lot.

vmatare, 2021-11-17:
Although it's a different error it might be the same issue as in #165. You can check the module options like this:
```
cat /sys/module/thinkpad_acpi/parameter/fan_control
```
I typed the path from memory, so it's unverified. Look around if it's incorrect, errors are more likely towards the end ;-)

koutheir, 2021-11-17:
The actual path is `/sys/module/thinkpad_acpi/parameters/fan_control`.

salvoc81, 2021-11-18:
Hello @vmatare and @koutheir   
The content of that file is the same for me.  
```
$ cat /sys/module/thinkpad_acpi/parameters/fan_control
Y
```  
And following are the permissions
```
$ exa -l /sys/module/thinkpad_acpi/parameters/fan_control
.r--r--r-- 4.1k root 18 11月 10:16 /sys/module/thinkpad_acpi/parameters/fan_control
```

Do you have the same permission for this file?

koutheir, 2021-11-18:
> Do you have the same permission for this file?

Yes:
```shell
$ ls -lh /sys/module/thinkpad_acpi/parameters/fan_control
-r--r--r-- 1 root root 4.0K Nov 18 09:24 /sys/module/thinkpad_acpi/parameters/fan_control
```

vmatare, 2021-11-18:
oh sorry, it's a very simple error in using `sudo`. The file redirection takes place before the `sudo`, i.e. when you do `sudo echo bla > something`, the file `something` is opened with your user ID, not as root.

koutheir, 2021-11-18:
My bad, I just read your original question. Let me explain what's actually going on.
```
$ sudo echo level 1 > /proc/acpi/ibm/fan
  ^    ^    ^     ^ ^ ^
  1    2    3     4 5 6
```

The line above is executed as follows **by your shell**:
1. Open `/proc/acpi/ibm/fan` (because 6) for writing and set it as the standard output (because 5) of `sudo` (because 1).
2. Run `sudo` with arguments 2, 3 and 4.

The command 2, 3 and 4 will run in privileged mode thanks to `sudo`. That is probably what you expect.

However, the file `/proc/acpi/ibm/fan` will be open for writing **by your shell**, not by `sudo`. Because your shell is unprivileged, that file cannot be open for writing, and your shell receives the `Permission denied` error due to that, and step (1) above fails.

There are multiple ways to fix this, all of them try to ensure that the command inside `sudo` is the one that opens the file for writing. You can use the command `tee` for this purpose, or create another shell that runs the command inside the `sudo` command, or use `su` instead. I personally prefer using `tee` for this.

closed 2022-08-12 by vmatare