vmatare/thinkfan #169: Thinkfan does not work with the latest update of `yaml-cpp.0.7.0`

javadr, 2021-11-20:
I updated my Linux distro (Manjaro) to the latest version, today. After that, `thinkfan` didn't work.
```
sudo systemctl status thinkfan
× thinkfan.service - simple and lightweight fan control program
Loaded: loaded (/usr/lib/systemd/system/thinkfan.service; enabled; vendor preset: disabled)
Drop-In: /etc/systemd/system/thinkfan.service.d
└─override.conf
Active: failed (Result: exit-code) since Sat 2021-11-20 14:30:55 +0330; 4s ago
Process: 39098 ExecStart=/usr/bin/thinkfan $THINKFAN_ARGS (code=exited, status=127)
CPU: 1ms

Nov 20 14:30:55 linuxlaptop systemd[1]: Starting simple and lightweight fan control program...
Nov 20 14:30:55 linuxlaptop thinkfan[39098]: /usr/bin/thinkfan: error while loading shared libraries: libyaml-cpp.so.0.>
Nov 20 14:30:55 linuxlaptop systemd[1]: thinkfan.service: Control process exited, code=exited, status=127/n/a
Nov 20 14:30:55 linuxlaptop systemd[1]: thinkfan.service: Failed with result 'exit-code'.
Nov 20 14:30:55 linuxlaptop systemd[1]: Failed to start simple and lightweight fan control program.
```
And trying `thinkfan -b` in a shell produced the following error:
```
thinkfan: error while loading shared libraries: libyaml-cpp.so.0.6: cannot open shared object file: No such file or directory
```

As far as I see, it depends on `yaml-cpp-0.6.0`, however, the latest version on my machine is [`yaml-cpp-0.7.0`](https://github.com/jbeder/yaml-cpp/releases/tag/yaml-cpp-0.7.0), now.  So, to resolve this issue I made a symbolic link to the new one with the old name. 
```
sudo ln -s /usr/lib/libyaml-cpp.so.0.7.0 /usr/lib/libyaml-cpp.so.0.6
```
It works for me now, but I don't know whether I am on the right path or not.

koutheir, 2021-11-20:
> It works for me now, but I don't know whether I am on the right path or not.

This is very wrong, and it is against the reasons versions were invented. Applications using that library through the 0.6 symbolic link (including `thinkfan` in this instance) might fail in very mysterious ways.

The correct solution is to recompile `thinkfan` from sources. Either on your computer, or on a server farm that would provide the recompiled executable.

If `thinkfan` depends specifically on version 0.6, then `thinkfan` source code needs to be updated, or `thinkfan` needs to be **statically** linked against the version 0.6 of the library.

koutheir, 2021-11-20:
As far as I could figure out, `thinkfan` should be compatible with version 0.7 of `libyaml-cpp`. This means that you should solve your issue by recompiling the `thinkfan` binary from source. If you installed `thinkfan` from ArchLinux AUR, then uninstalling and reinstalling `thinkfan` should solve the issue for you.

javadr, 2021-11-20:
> As far as I could figure out, `thinkfan` should be compatible with version 0.7 of `libyaml-cpp`. This means that you should solve your issue by recompiling the `thinkfan` binary from source. If you [...]

I did uninstall the `thinkfan` and reinstalled it from AUR, but it didn't work for me and I got the same error!

vmatare, 2021-11-23:
this is a duplicate of #162. Ping @bssb, maybe there is really something wrong with that _AUR helper_ that's supposed to detect & fix these situations?

vmatare, 2021-11-23:
So @javadr on the referenced bug I wrote that this is most likely a bug in your distribution package, but its maintainer @bssb assures me that all dependencies are represented correctly there. So that would mean it's a bug in your distribution's dependency manager.

wiwie, 2021-12-03:
reinstall with cleanBuild did solve the problem for me.

javadr, 2021-12-05:
Thanks @wiwie. Actually, reinstalling with cleanBuild solved the issue.

closed 2021-12-05 by javadr