vmatare/thinkfan #172: Version 1.3.0 builds 1.2.2

eldorel, 2021-12-03:
I initially noticed that the version in AUR is building 1.2.2, but after grabbing the release zip file and building it I'm seeing the same issue. 

> $ thinkfan -h
> thinkfan 1.2.2: A minimalist fan control program 
> ...clipped...

Man page from AUR version:

[image]

I'm also getting some weird hwmon file permission errors while running as root, but they may just be a symptom.

vmatare, 2021-12-05:
Yes, I forgot to increment the version number in the code with the 1.3.0 release. I'm thinking about what I can do to prevent that from happening, but it's not trivial. However I can tell you that it's really just the version number. If you built the 1.3.0 codebase, you have the 1.3.0 release, it's just reporting a wrong version number.

And that would also mean that whatever weird permission problem you're seeing may definitely be worth an issue report ;-)

vmatare, 2022-03-29:
Fixed with the 1.3.1 release and should hopefully be less likely to happen now :-/

closed 2022-03-29 by vmatare