vmatare/thinkfan #174: Improve support for discrete GPUs under hwmon

atrueba, 2021-12-27:
Hi,

I'm using thinkfan 1.2.1 (Debian testing) on a Thinkpad P14s. This device is equipped with a discrete Nvidia GPU and I'm currently using nouveau drivers.

With this setup, hwmon creates a temp sensor for the GPU at boot under "thinkpad" group, but the tempX_input file is only accesible while the discrete GPU is enabled via primus/pvkrun. The file is always present, but any file access while the GPU is off fails with "device or address invalid".

The problem is that if thinkfan is configured to read that sensor it'll fail to start, even if declared as "optional", as it detects the sensor, tries to read it and fails, so the only way to monitor that sensor is restarting thinkfan while the GPU is enabled.

I suggest to modify the "optional" behaviour to include this case: if a sensor is declared optional but the given temp file is inaccesible just ignore the error and continue with all other sensors available. When the GPU is actually enabled the system will automatically make that reading available and thinkfan can use it as if it were a hotplug event.

vmatare, 2021-12-28:
This is probably a duplicate of https://github.com/vmatare/thinkfan/issues/109, which was fixed in v1.2.2.

vmatare, 2022-03-29:
May also be the same as #180, or maybe both.

vmatare, 2022-03-29:
Closing this in favor of the two more specific issues. When both are fixed, this should be fully adressed, as well.

closed 2022-03-29 by vmatare