vmatare/thinkfan #176: enhancement: option to smooth temperature data

jdchristensen, 2022-01-23:
On my X1 Extreme Gen 4, the CPU package temp and core temps often briefly spike up, but then immediately drop down.  This causes thinkfan to turn the fans on, but by the time they ramp up, the spike is already gone.  I know about the bias option, but the spikes are too large to be managed by it without compromising desired increases.  I propose adding an option `-w weight` which smooths things out by forming a weighted average of the new reading with the old accumulated value.  Here `weight` is a floating point number between 0 and 1, and we would calculate:
```
  newvalue = (1.0-weight)*oldvalue + weight*newsensorreading
```
When `weight=1.0`, we just use the newsensorreading, as before, but when `weight=0.7`, say, we would smooth it out by averaging against the previous accumulated value (not the previous raw reading).

IgnatBeresnev, 2022-02-04:
Can confirm the spikes on P1 Gen 4 (i7-11800H) as well

Frank-Steiner, 2022-03-15:
I came here to ask for sth. similar and found this :) So I second such an option, seeing such spikes on my Lenovo T580 with i7-8550U due to the TurboBoost behaviour.

vmatare, 2022-03-29:
It's certainly an interesting proposal, but the upcoming 2.0 release is pretty full already with some long-standing bugfixes and another major documentation overhaul for multi-fan support. So I think I'll take a closer look at this for 2.1...

vmatare, 2022-03-29:
Just thought a little more about @jdchristensen's statement that the `-b` option doesn't have the desired effect. I'm entirely sure what kind of temperature changes we're talking about here and what's actually desired there and why. Since you say "spikes", I'm thinking about short and sudden increases as they are often reported by modern Intel processors, so let's say you have:

`50, 51, 65, 78, 60, 62`

With `-b-6` you arrive at:

`50, 51, 57, 61, 60, 61`

effectively canceling out the spike. With `-b-3` you'd get:

`50, 51, 61, 71, 60, 61`

Which would keep some of the spike. In fact, I'd consider the behavior with `-b-6` more desirable since in reality you don't care about the spike at all. It's just completely absorbed by the heatsink, which has a heat capacity that's probably some orders of magnitude greater than that of the tiny little DIE.

So I guess my question is: Why not use something like `-b-6` or even `-b-8` to effectively ignore everything except slow and consistent heatup?

vmatare, 2022-03-29:
Also note that in modern laptops the fan's heat exchanger never sits directly in the CPU's heatsink, but is connected to it with a heatpipe that further increases heat capacity and delays the effectiveness of active cooling. So unless I'm overlooking something fundamental, I'm pretty certain there's really no point in reacting at all if temperature drop down again before we even started reacting.

jdchristensen, 2022-03-30:
@vmatare I agree that the goal is to not turn the fan on at all when the spike is brief.  But I don't understand the example bias calculations you did, and how they result from the formula `current_tmax = current_tmax + delta_t * BIAS / 10`.  Can you explain?

Also, the spikes I'm talking about are much more dramatic.  At one second intervals, I'll see temps like 39, 41, 39, 95, 40, 39.  It's that 95 that I want to ignore, and so the amount of bias needed would be huge.

vmatare, 2022-03-30:
Yeah, when calculating the examples I noticed that the formula in the manpage doesn't describe the whole truth. The actual behavior is a little more complex. However in the simple example you gave, it works just as the formula says, e.g. for `-b-9` you'd get:

```
real temps:   39, 41, 39, 95, 40, 39
biased temps: 39, 41, 39, 45, 40, 39
```
So the only interesting part is the jump from 39 to 95, i.e.
- `current_tmax` = 95
- `delta_t` = 95 - 39 = 56
- `BIAS` = -9
So with the formula from the man page: `current_tmax = 95 + 56 * -0.9 = 44.6 =~ 45`

HOWEVER:
- If a temperature drops even by -1 °C, the bias for that sensor is eliminated immediately, so if you had e.g. 39, 95, 94, 40, with `-b-9` you'd get 39, 45, 94, 40. Maybe that sometimes creates sudden and undesired fan speedups.
- If delta_temp stays between 0 and +2 °C, the `BIAS` factor is slowly decreased with each loop until it reaches 0. Maybe this is the essential bit of missing information which had you thinking that biasing won't do what you want?

The behavior is this weird because it was designed to also allow for exaggeration of temperature increases by giving a positive bias. Nowadays this seems to be getting rarer, but older CPUs could have a very slow load-to-temperature response.

So most importantly: have you tried running thinkfan with e.g. `-b-8` and actually noticed any undesirable behavior? And if you didn't I'd be interested to know what else you may have found lacking in the docs (beside the incomplete biasing logic description).

vmatare, 2022-03-30:
btw, I also noticed an error in the example I gave above. The actual behavir should be more like this:

With `-b-6`:
```
real temps:   50, 51, 65, 78, 60, 62
biased temps: 50, 51, 57, 70, 60, 62
```

With `-b-8`:
```
real temps:   50, 51, 65, 78, 60, 62
biased temps: 50, 51, 54, 68, 60, 62
```

I guess it's pretty obvious the documentation for the `-b` option could use an overhaul ;-)

IgnatBeresnev, 2022-03-30:
If this helps in any way, I can collect some data, for instance append CPU temp and fan speed every second to a txt file. Maybe that'll give some insight into the spikes.

If you want me to run some specific command, let me know, will do.

n0x32schneider, 2025-04-07:
Has anything happened regarding this topic since it was last discussed?
[image]

Here for example I run with -b-9, huge spike gets offset nicely: 47 (0) -> 78(-25) 
Then dropdown delta gets applied negatively, although it should be reset on falling temps? 78(-25) -> 48(-19) ... 
Either -25 to -19 is decay and not correctly reset, or the offset is basically applied in the wrong direction, as the 78 should have been an ignore, but gets included in bias calculation. This oscillates my fan levels, as the undershoot happens because of the huge spike earlier. 
And then follows slow decay of bias 48(-19) -> 47(-14) -> 46(-4) which seems reasonable, and would at least change fan levels slowly. 

So, all in all not very intuitive to use, and I can not eliminate unwanted fan level changes. Some smoothing function, maybe windowed over the last X values as well would be way simpler and less error prone.

What we want to see is a real increase in temperature. So we could de-bounce it -> last X readings above threshold. And the threshold is some reasonable (configurable?) delta around the latest read temperature.
Plus we calculate the estimated temperature with weighted average.
If the debounce notices a real temperature spike, maybe increase the weight of the new incoming measurements,
so the average value gets less weight, the new higher temperature more weight, if the debounce shows the readings might be not just one spike.