vmatare/thinkfan #179: enhancement: option to delay the next lower level

Frank-Steiner, 2022-03-15:
Hi, this is a proposal for an enhancement, little bit related to the "smooth temperature data" enhancement proposal.

Due to the turbo boost option of my i7-8550U (Lenovo T580 laptop) the cpu temperature jumps high and low very quickly and often (e.g. from 48°C up to 65-70, stay there for some seconds, drop back to 48). I cannot ignore such high temperature for more than 1-2 seconds, so the fan starts and even might climb two levels, then fall back after a few seconds.

In my situation it would help a lot if there was an option to delay entering the next lower fan level for some time. It would sound a lot better (and likely also be better for the fan) if I could e.g. specify "stay at least 30 seconds in fan level "1" before going to level "off". 

Because then fan wouldn't go on/off/on/off... but stay in level 1 or 2 as long as I e.g. open the next product page on Amazon in Firefox (which always brings the fan up and turns if down again) frequently.

Due to the large differences (48 to 70 and back to 48) such a behaviour cannot be reached by overlapping limits.

Either a global delay option or an extender syntax for a per-level setting like "[0, 0, 52, 20]" for "stay at least 20 seconds in this level before lowering the level" would be helpful.

Apart from that: great script!  Many thanks for my now quite quiet (and still cool) laptop :-)

vmatare, 2022-03-29:
Since you looked at the other issue already I'm guessing you already tried running thinkfan with a biasing factor of -10 (`thinkfan -b-10`)? That should cause it to ramp up more slowly and therefore at least alleviate the sudden & drastic fan speed changes.

Frank-Steiner, 2022-03-31:
Yes, I tried with the bias, but it's not possible to prevent the fan going up because the spikes aren't super short. Instead the temps are going high for several seconds and maybe changing from 70 to 75, back to 68, up to 72 again etc. It's not possible to ignore them unless I would also set -s really high. And it's ok that the fan starts in these situations.
But when I e.g. browse a webpage with several subpages which I open, glance through for 10-20 seconds, then go back, open the next etc., the fan will always start when loading those pages (because of the many elements in them), then stop again after 5-10 seconds, then start again after 10 seconds for the next page etc. And that really is much more disturbing as if the fan would keep running at least in the lowest level. 

I guess this situation is very special, but I hit it quite often when browsing webpages. Thus, a possibility to delay the return to the next lower level would help here. 
That's actually what the Thinkpad laptops do when using level "auto". Whenever the fan has gone up it will go down quite delayed so that in the browsing example the fan would never go below level 2. When I quit the browser while fan is at level 5 and the laptop turns idle (cpu temp dropping back to 48 after 5 seconds), it will take like 5 seconds to go back to level 4, 15 seconds to 3, 30 seconds to 2, and then 1-2 minutes to fall back to 1. Turning off might take 5 minutes or more. These delays are ok for high levels, but way to long for lower levels...