vmatare/thinkfan #180: "optional" keyword not working?

pblack476, 2022-03-29:
I have an optimus enabled laptop and thinkfan only works when I engage my nVidia card. When I boot using just the iGPU `thinkfan status` reports `no nvml device found` which indicates that it's exiting when searching for a nvml device that is not enabled.

I have used the _optional_ keyword in the nvml sensor (as below) but it does not seem to ignore it if its not found.

Not a huge issue since I don't really need thinkfan when using iGPU only, but still would like the option.

Here is my thinkfan.yaml:

`sensors:
  - hwmon: /sys/devices/platform/thinkpad_hwmon/hwmon
    name: thinkpad
    indices: [1, 2, 3, 4, 5, 6, 7, 8]
    optional: true
    
  - hwmon: /sys/devices/platform/coretemp.0/hwmon
    name: coretemp
    indices: [1, 2, 3, 4, 5, 6, 7]
    optional: true
    
  - nvml: 01:00.0
    optional: true    

fans:
  - tpacpi: /proc/acpi/ibm/fan

levels:
  - [0, 0, 50]
  - [1, 50, 54]
  - [2, 54, 58]
  - [3, 58, 60]
  - [4, 60, 63]
  - [5, 63, 65]
  - [6, 65, 70]          
  - ["level full-speed", 70, 255]           
`

vmatare, 2022-03-29:
yeah, in the current thinkfan releases, the `optional` keyword doesn't affect startup. So when thinkfan starts for the first time, all sensors have to be available, including the `optional` ones. However, a fix for that is currently in development in the https://github.com/vmatare/thinkfan/tree/init_retries branch. You can go try it if you want to help out. It's currently in a "works for me" state, i.e. I'm running it on my daily system, but there's probably still some bugs in there that I haven't noticed yet and also some refinements & cleanup I need to do before merging it into the `master` branch.

vmatare, 2022-08-13:
@pblack476 This should be fixed in the current `master` now. I'd appreciate a notice if you find that it works for you now ;-)

pblack476, 2022-08-13:
It works now.

closed 2022-08-14 by vmatare