vmatare/thinkfan #185: no pwm1, but pwm1_enable exists on laptop

ImALoooserOk, 2022-04-28:
sorry if this is a stupid thing to make an issue about, but as I was configuring thinkfan for the first time, I realized that I have a `pwm1_enable`, but no `pwm1` in `/sys/class/hwmon/hwmon#/`, is this normal? did I miss a step?

vmatare, 2022-07-27:
Hi @ImALoooserOk, sorry about the late reply. I've heard about some drivers with limited support for userspace fan control. First of all I'd need to know which hwmon driver we're talking about. You should find it by looking where the `device/driver` symlink points in your hwmon directory.

As a wild guess, if you're talking about the `thinkpad_acpi` driver, you might need to load with the option `fan_control=1`.

vmatare, 2023-04-01:
Assuming module options solved this. Feel free to reopen if needed.

closed 2023-04-01 by vmatare