vmatare/thinkfan #186: thinkfan not working on P14s gen2 AMD thinkpad

sebastian-wien, 2022-05-03:
hey vmatare!

thinkfan is not starting on the new thinkpad P14s Gen2 with AMD Ryzen 5000 series CPU.

i came across your tool through https://blog.monosoul.dev/2021/10/17/how-to-control-thinkpad-p14s-fan-speed-in-linux/

my config:
```sensors:
  # GPU
  - tpacpi: /proc/acpi/ibm/thermal
    indices: [1]
  # CPU
  - hwmon: /sys/class/hwmon
    name: coretemp
    indices: [2, 3, 4, 5]
  # Chassis
  - hwmon: /sys/class/hwmon
    name: thinkpad
    indices: [3, 5, 6, 7]
  # SSD
  - hwmon: /sys/class/hwmon
    name: nvme
    indices: [1, 2, 3]
    correction: [-5, 0, 0]
  # MB
  - hwmon: /sys/class/hwmon
    name: acpitz
    indices: [1]

fans:
  - tpacpi: /proc/acpi/ibm/fan

levels:
  - [0, 0, 60]
  - [1, 58, 70]
  - [2, 68, 75]
  - [3, 73, 80]
  - [4, 78, 85]
  - [5, 83, 87]
  - [6, 85, 90]
  - [7, 88, 95]
  - ["level full-speed", 93, 32767]
```
leading to this error:

> ⋊> ~ thinkfan -c /etc/thinkfan.conf -v                                                                                              
>ERROR: /etc/thinkfan.conf:7:
>    name: coretemp
>          ^
> Could not find a hwmon with this name.

so i removed the name, which is leading to this error:

> ⋊> ~ thinkfan -c /etc/thinkfan.conf -v                                                                                              
> ERROR: Could not find an `hwmon*' directory or `temp*_input' file in /sys/class/hwmon/hwmon0/device/device.

https://github.com/vmatare/thinkfan/blob/7ba83b93c48edaeb40ebf0086fc285af74f2eb90/src/hwmon.cpp#L230

is it because the /sys/class/hwmon directory is completely scanned either way, or because i commented out the name for the CPU hwmon section in the config file? or the new AMD architecture?

vmatare, 2022-05-04:
Hi @sebastian-wien,
> is it because the /sys/class/hwmon directory is completely scanned either way, or because i commented out the name for the CPU hwmon section in the config file? or the new AMD architecture?

Normally the search should terminate wherever a `temp*_input` file is found. The error message you're getting may be a bit misleading. To find out what's going on, take a look at what you actually have in `/sys/class/hwmon/hwmon0`, i.e. do
```
ls -la /sys/class/hwmon/hwmon0
```
That's where your temperature input files should normally be found. My guess is that the indices you specified in your config simply don't match what you have in there.

Incidentally, I'm currently rewriting this exact part of the config & search logic. I've just pushed the latest development snapshot to the current master branch. If you want to help out, you can go ahead and try it out. I'd be very interested to know whether the current master version gives you a more helpful error message ;-)

vmatare, 2022-05-04:
Oh and one obvious thing: You definitely won't find a `coretemp` sensor on an AMD system. `coretemp` is specific to Intel. Just look around in `/sys/class/hwmon` or run `sensors` as root to find out what sensors/fans you do have. However I'd still appreciate it very much if you could test your broken config with the current master version.

sebastian-wien, 2022-05-06:
awesome, thanks for your reply!
it seems that it was all about the name: as soon as i changed it from `coretemp` to `thinkpad` the error vanished. +1

the tool seems to work, systemctl status is green, but i was wondering if it works correctly, because: when the first level of the fan is activated (>65° -> `- [0, 0, 65]`) the fan instantly ramps up to 3600rpm (according to `sensors`). is there something i can do about?

vmatare, 2022-05-06:
> the tool seems to work, systemctl status is green, but i was wondering if it works correctly, because: when the first level of the fan is activated (>65° -> `- [0, 0, 65]`) the fan instantly ramps up to 3600rpm (according to `sensors`). is there something i can do about?

If you want to observe thinkfan's operation, just follow the syslog:
```
systemctl -elf -u thinkfan
```
About the fan RPM, there's probably not much you can do. On all thinkpads I've seen the embedded controller limits the number of available fan speeds to 7, even if you access it through hwmon/pwm which in theory ranges from 0 to 255. As far as I know, there's no way for the user to change the RPM of a given fan level. If level 1 already seems loud for you, the best you can do is to allow good passive cooling and quit programs that waste CPU power in the background. It might also help to let it cool down further than 58 °C, or you might even let level 2 reach down below 60°C if that gets temperatures down faster so the fan turns off sooner. Oh and sometimes, automatic fan control is a little more fine-grained than the manual levels, so you might also want to use that in certain temperature ranges, e.g. something like this:

```yaml
levels:
  - [0, 0, 64]
  - ["level auto", 54, 80]
  - [5, 75, 80]
  - [6, 78, 90]
  - [7, 88, 95]
  - ["level full-speed", 93, 32767]
```

closed 2022-05-07 by sebastian-wien