vmatare/thinkfan #189: Error after boot: /proc/acpi/ibm/fan: No such file or directory

binarious, 2022-06-17:
**Steps to reproduce**
- Boot with thinkfan service enabled with:
  - ThinkPad T14 Gen. 2 Intel 
  - Kernel 5.15.46
  - thinkfan 1.3.1-1

**Expected behavior**
Thinkfan service is running without errors.

**Actual behavior**
Thinkfan fails with the following error:
```
systemd[1]: Starting simple and lightweight fan control program...
thinkfan[552]: ERROR: ~TpFanDriver: Resetting fan control in /proc/acpi/ibm/fan: No such file or directory
thinkfan[552]: ERROR: ~TpFanDriver: Resetting fan control in /proc/acpi/ibm/fan: No such file or directory
thinkfan[552]: ERROR: init: Initializing fan control in /proc/acpi/ibm/fan: No such file or directory
systemd[1]: thinkfan.service: Control process exited, code=exited, status=1/FAILURE
```

After `sudo systemctl restart thinkfan.service` it starts working again:
```
systemd[1]: Starting simple and lightweight fan control program...
thinkfan[15446]: Daemon PID: 15447
systemd[1]: Started simple and lightweight fan control program.
```

**thinkfan.yaml**
```
sensors:
  - hwmon: /sys/class/hwmon
    name: thinkpad
    indices: [1]

fans:
  - tpacpi: /proc/acpi/ibm/fan

levels:
  - [0, 0, 50]
  - [1, 45, 65]
  - ["level auto", 60, 85]
  - ["level disengaged", 80, 255]
```

**Additional details**
- Installed from the AUR package `thinkfan`.
- Service file: https://aur.archlinux.org/cgit/aur.git/tree/thinkfan.service?h=thinkfan

koutheir, 2022-06-17:
It looks like `thinkfan` is started too early, before `/proc/acpi/ibm/fan` is available.

binarious, 2022-06-17:
I guess so. Maybe the `After=lm_sensors.service` in the service file of the AUR is wrong/incomplete, but I'm not sure. What service does provide `/proc/acpi/ibm/fan` so that it should be waited for?

koutheir, 2022-06-17:
`/proc/acpi/ibm/fan` is [provided by `thinkpad_acpi`](https://wiki.gentoo.org/wiki/Fan_speed_control/thinkfan), which is a module, not a service.

I'm not sure what service would make sense for `After=`. Maybe delaying the start up of `thinkfan` by a few seconds would make sense in this situation?

vmatare, 2022-07-26:
Hi @binarious, the latest `master` version supports a new config setting `max_errors: NUM` that will let thinkfan retry the initialization of a given fan or sensor `NUM` times. Alternatively, you can mark a fan/sensor with `optional: true` to allow retrying indefinitely. So if you want to help testing this new feature (which is slated for the 2.0 release), you can try something like this:
```yaml
sensors:
  - hwmon: /sys/class/hwmon
    name: thinkpad
    indices: [1]
    max_errors: 10

fans:
  - tpacpi: /proc/acpi/ibm/fan
    max_errors: 10

levels:
  - [0, 0, 50]
  - [1, 45, 65]
  - ["level auto", 60, 85]
  - ["level disengaged", 80, 255]
```
If a fan or sensor is unavailable, it is retried every loop, so if your're using the default sleep time of 5 seconds, that should give you ~50 seconds of retries before thinkfan fails to start.

binarious, 2022-08-08:
Hi @vmatare, confirmed fixed. Thank you!

closed 2022-08-08 by binarious

jeandestouches, 2023-02-09:
I faced the same issue with thinkfan 1.3.1 on Debian bookworm and kernel 6.1.10.
/proc/acpi/ibm/fan was not available
so I added a thinkfan.timer until the new thinkfan version is available.

/etc/systemd/system/thinkfan.timer

```
[Unit]
Description=Delaying thinkfan start

[Timer]
OnStartupSec=20
AccuracySec=5```

```sudo systemctl disable thinkfan.service
sudo systemctl add-wants multi-user.target thinkfan.timer
sudo systemctl daemon-reload```

Now it is starting correctly.