vmatare/thinkfan #190: Suggestions for more clarifying instructions in readme

emilianomacht, 2022-06-20:


mayonezo, 2022-08-18:
Well, this would have helped me not spend 3 hours searching the web.

vmatare, 2022-12-02:
Thanks for the feedback. So, (1) is in the README now and (3) has been incorporated into the appropriate error message. What exactly went wrong with (4) I can't tell because for that I'd need the literal config and the exact (and complete!) output with the `invalid sensor entry` message. I'm ignoring (2) here because that's a Debian-specific thing that Debian users should know, and thinkfan is not a Debian handbook.

Since multiple people seem to have experienced a problem with `- chip: thinkpad-isa-0000`, maybe someone can get me the exact config with the exact error output?