vmatare/thinkfan #191: Issue when applying mode to pwm

berrabe, 2022-07-07:
### ISSUE
---
IDK why, but thinkfan on my laptop do weird things. Thinkfan success read temp and choose right level mode, but when choosen mode applied to fan conf file, it's wrong. For example for temp under 50 °C set to level 1, thinkfan know that and show level 1 (3.xxx RPM) on log, but the reality on fan site is level 0 (0 RPM)

&nbsp;

### PoC
---
tested on **ThinkPad X240** with **manjaro 21.3.2 (ruah)**, and **thinkfan 1.3.1 (AUR)**

here's my thinkfan conf file

```yml
# created by berrabe (ThinkPad X240)
# references:
#   - minimal_conf: |
#     sed -E 's/^ *#.*$//g; /^$/d' /usr/share/doc/thinkfan/examples/thinkfan.yaml
#   - docs: |
#     https://wiki.archlinux.org/title/fan_speed_control#ThinkPad_laptops
#     man thinkpad.conf
#     thinkfan github issue 45 and 146
---
sensors:
  - hwmon: /sys/class/hwmon/hwmon3/temp1_input
  # - hwmon: /sys/class/hwmon
  #   name: thinkpad
  #   indices: [1]
fans:
  # - hwmon: /sys/class/hwmon/hwmon3/pwm1
  - tpacpi: /proc/acpi/ibm/fan
levels:
  - [0, 0, 40]
  - [1, 40, 50]
  - ["level auto", 50, 100]
```

and tested with `thinkfan -n`

[image]

vmatare, 2022-07-17:
It appears you're using `/proc/acpi/ibm/fan`, but you're showing the contents of `/sys/class/hwmon/hwmon3/pwm1*`. Could you please show me the contents of `/proc/acpi/ibm/fan` when this happens?

Also, what happens if you actually do use `/sys/class/hwmon/hwmon3/pwm1` as fan control?

vmatare, 2022-11-20:
closing for lack of feedback. Feel free to reopen if new information comes up.

closed 2022-11-20 by vmatare