vmatare/thinkfan #193: Setting fan to 7 seems to set computer into a limited power state?

kylebakerio, 2022-07-14:
I have an x1 extreme gen 4. I'm trying to benchmark my CPU before a repaste. As part of getting a consistent benchmark, I'd like a consistent fan speed, instead of just relying on auto, which does not seem to bring the fans up to even 50% of their potential.

So I installed thinkfan. When I set the fan speed to 7, the fans do indeed get fast, and the processor does indeed get much colder. But my actual speed was significantly reduced. Setting the fans back to auto and stressing the processor again, I now see that my processor is in a 80c/2.6ghz mode, equivalent to a 'battery saver' configuration, and I cannot seem to set it to `high power mode` again without restarting the computer.

What's causing this? How can I get full speed fans without limiting my processor?

kylebakerio, 2022-07-14:
more detailed info here: https://www.reddit.com/r/linuxhardware/comments/vyjg6x/turning_on_fan_to_max_with_thinkfan_seems_to_set/

nothing seems to work, and restarting computer doesn't undo this. It's like my OS is permanently in `balanced` mode now... How is this possible?

vmatare, 2022-07-17:
Well, that sounds like a slightly excessive safety measure on Lenovo's part, which I must say isn't entirely unexpected. It's entirely possible for them to limit the CPU freq when you (or rather thinkfan) enable(s) manual (i.e. software) fan control. These kind of shenanigans are why I don't buy these super-slim thingies. There's simply not enough thermal flow for them to be able to sustain full load over longer periods of time. Put in a high-powered CPU and you have to implement some serious constraints to limit the number of unsatisfied customers who somehow managed to break their stuff.

This is also the reason I put big fat warning labels all over thinkfan, but maybe there are still enogh dorks out there who misconfigure things and then go complain to Lenovo ;-)

kylebakerio, 2022-07-20:
So, after many hours tinkering, I don't know where or how it started, but I got it resolved, and it seems like it must have been a coincidence and not truly thinkfan's fault. I was able to get max fans going and a 95c limit enforced to open it up and get true benchmarks. All details in the linked reddit post above.

closed 2022-07-20 by kylebakerio