vmatare/thinkfan #194: Problem with building thinkfan, DSO missing from command line

jorikoolstra, 2022-08-22:
As thinkfan doesn't come packaged with my Debian versions I tried to build it. I installed all the libraries as instructed in the README file, and followed the buildd instruction. Running cmake works fine, but then when I run make I get the following linker error that I cannot solve no matter what I try (this is actually my first time building a project):

`/usr/bin/ld: CMakeFiles/thinkfan.dir/src/parser.cpp.o: undefined reference to symbol '_ZTVN10__cxxabiv121__vmi_class_type_infoE@@CXXABI_1.3'
/usr/bin/ld: /lib/x86_64-linux-gnu/libstdc++.so.6: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/thinkfan.dir/build.make:269: thinkfan] Error 1
make[1]: *** [CMakeFiles/Makefile2:95: CMakeFiles/thinkfan.dir/all] Error 2
make: *** [Makefile:149: all] Error 2`

I don't have the expertise to know whether I screwed something up or whether this is a genuine issue. I would really appreciate help!

vmatare, 2022-08-23:
Hm, from the log lines you posted I can't really tell what might have gone wrong. It looks like something is wrong with your gcc/g++ setup. If you post the full output of this we might find something:
```bash
make clean
VERBOSE=1 make
```

LATENTBOUNCE, 2022-08-24:
Thank-you so much for your help! OK, that was quite interesting! I see a potential issue, namely that cmake tries to use gcc for building, instead of g++. Maybe that doesn't work on my system? It doesn't explicitly link libstdc++6 either.
(also, unrelated, where should I really have my build directory for thinkfan, now I just have a build directory in source? this is not the place where thinkfan is installed, correct? just build? is there a designated place for build files?).

See attachment for logs.
[log.txt](https://github.com/vmatare/thinkfan/files/9411257/log.txt)

vmatare, 2022-08-24:
> I see a potential issue, namely that cmake tries to use gcc for building, instead of g++. Maybe that doesn't work on my system? It doesn't explicitly link libstdc++6 either.
You're exactly right.

> (also, unrelated, where should I really have my build directory for thinkfan, now I just have a /build directory in /usr/local/src/thinkfan? this is not the place where thinkfan is installed, correct? just build? is there a designated place for temporary build files?).

`thinkfan/build` is fine. It puts all binaries there, ready to be installed to `${CMAKE_INSTALL_PREFIX}` with `make install`.

vmatare, 2022-08-24:
Just [updated the readme](https://github.com/vmatare/thinkfan/commit/11ce0006533e8344da83703f8821a854e578bb4d) with some setup instructions for debian derivates.

LATENTBOUNCE, 2022-08-24:
Thanks! However, I had already installed g++ to be sure, and cmake still doesn't use it. Do you know how to force cmake to use it or why it isn't using it in the first place?

vmatare, 2022-08-25:
I'm not sure what is confusing cmake here. If I test it with a plain Debian Bullseye docker container where I install these exact packages it works correctly.

I've made a small change on the master branch that should tell CMake to use C++ for everything. Would be great if you could try and see if it helps.

vmatare, 2023-04-01:
Closing as it appears solved. Feel free to reopen if you observe this again.

closed 2023-04-01 by vmatare