vmatare/thinkfan #195: yaml-cpp undefined reference

mnissov, 2022-08-24:
Attempting to build release 1.3 on PopOS 20.04 results in several `undefined reference to` errors with `yaml-cpp`.

Does `thinkfan` require a specific `yaml-cpp` version? In this release it just makes a check against versions older than 0.5.3

## Output from cmake:
```
cmake -D CMAKE_BUILD_TYPE=RelWithDebInfo -D BUILD_TESTS=TRUE ..
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Checking for module 'systemd'
--   Found systemd, version 245
-- Checking for module 'openrc'
--   No package 'openrc' found
-- Checking for module 'yaml-cpp'
--   Found yaml-cpp, version 0.7.0
-- Checking for module 'libatasmart'
--   No package 'libatasmart' found
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    BUILD_TESTS

-- Build files have been written to: /home/morten/repos/thinkfan/build
```
## Output from `make`
```
make
Scanning dependencies of target thinkfan
[ 11%] Building CXX object CMakeFiles/thinkfan.dir/src/thinkfan.cpp.o
[ 22%] Building CXX object CMakeFiles/thinkfan.dir/src/config.cpp.o
[ 33%] Building CXX object CMakeFiles/thinkfan.dir/src/fans.cpp.o
[ 44%] Building CXX object CMakeFiles/thinkfan.dir/src/sensors.cpp.o
[ 55%] Building CXX object CMakeFiles/thinkfan.dir/src/message.cpp.o
[ 66%] Building CXX object CMakeFiles/thinkfan.dir/src/parser.cpp.o
[ 77%] Building CXX object CMakeFiles/thinkfan.dir/src/error.cpp.o
[ 88%] Building CXX object CMakeFiles/thinkfan.dir/src/yamlconfig.cpp.o
[100%] Linking CXX executable thinkfan
/usr/bin/ld: CMakeFiles/thinkfan.dir/src/yamlconfig.cpp.o: in function `YAML::Node::Scalar[abi:cxx11]() const':
/usr/local/include/yaml-cpp/node/impl.h:169: undefined reference to `YAML::detail::node_data::empty_scalar[abi:cxx11]()'
/usr/bin/ld: /usr/local/include/yaml-cpp/node/impl.h:169: undefined reference to `YAML::detail::node_data::empty_scalar[abi:cxx11]()'
/usr/bin/ld: /usr/local/include/yaml-cpp/node/impl.h:169: undefined reference to `YAML::detail::node_data::empty_scalar[abi:cxx11]()'
/usr/bin/ld: /usr/local/include/yaml-cpp/node/impl.h:169: undefined reference to `YAML::detail::node_data::empty_scalar[abi:cxx11]()'
/usr/bin/ld: /usr/local/include/yaml-cpp/node/impl.h:169: undefined reference to `YAML::detail::node_data::empty_scalar[abi:cxx11]()'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/thinkfan.dir/build.make:189: thinkfan] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/thinkfan.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

vmatare, 2022-08-24:
I think I need to look at your actual compiler calls which are hidden in your log. Please run
```bash
make clean
VERBOSE=1 make
```
I also just updated the README with some setup instructions for debian derivates. Maybe they can also help you.

mnissov, 2022-08-25:
I've attached the output from making with `VERBOSE=1`. Regarding advice for debian derivatives, I've confirmed I have all the requisite packages, as said I can install thinkfan from apt it's just much older, except for `libsensors-yaml`. At the time of my checking there is nota pkg called `libsensors-dev`:
```
sudo apt install libsensors
libsensors4-dev               libsensors-applet-plugin-dev
libsensors5                   libsensors-config           
libsensors-applet-plugin0
```

```
VERBOSE=1 make
/usr/bin/cmake -S/home/morten/repos/thinkfan -B/home/morten/repos/thinkfan/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/morten/repos/thinkfan/build/CMakeFiles /home/morten/repos/thinkfan/build/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/morten/repos/thinkfan/build'
make -f CMakeFiles/thinkfan.dir/build.make CMakeFiles/thinkfan.dir/depend
make[2]: Entering directory '/home/morten/repos/thinkfan/build'
cd /home/morten/repos/thinkfan/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/morten/repos/thinkfan /home/morten/repos/thinkfan /home/morten/repos/thinkfan/build /home/morten/repos/thinkfan/build /home/morten/repos/thinkfan/build/CMakeFiles/thinkfan.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/morten/repos/thinkfan/build'
make -f CMakeFiles/thinkfan.dir/build.make CMakeFiles/thinkfan.dir/build
make[2]: Entering directory '/home/morten/repos/thinkfan/build'
[ 11%] Building CXX object CMakeFiles/thinkfan.dir/src/thinkfan.cpp.o
/usr/bin/c++  -DHAVE_SYSTEMD -DPID_FILE=\"/run/thinkfan.pid\" -DUSE_NVML -DUSE_YAML -DVERSION=\"1.3.1\" -I/home/morten/repos/thinkfan/include  -O2 -g -DNDEBUG   -Wall -std=c++1y -o CMakeFiles/thinkfan.dir/src/thinkfan.cpp.o -c /home/morten/repos/thinkfan/src/thinkfan.cpp
[ 22%] Building CXX object CMakeFiles/thinkfan.dir/src/config.cpp.o
/usr/bin/c++  -DHAVE_SYSTEMD -DPID_FILE=\"/run/thinkfan.pid\" -DUSE_NVML -DUSE_YAML -DVERSION=\"1.3.1\" -I/home/morten/repos/thinkfan/include  -O2 -g -DNDEBUG   -Wall -std=c++1y -o CMakeFiles/thinkfan.dir/src/config.cpp.o -c /home/morten/repos/thinkfan/src/config.cpp
[ 33%] Building CXX object CMakeFiles/thinkfan.dir/src/fans.cpp.o
/usr/bin/c++  -DHAVE_SYSTEMD -DPID_FILE=\"/run/thinkfan.pid\" -DUSE_NVML -DUSE_YAML -DVERSION=\"1.3.1\" -I/home/morten/repos/thinkfan/include  -O2 -g -DNDEBUG   -Wall -std=c++1y -o CMakeFiles/thinkfan.dir/src/fans.cpp.o -c /home/morten/repos/thinkfan/src/fans.cpp
[ 44%] Building CXX object CMakeFiles/thinkfan.dir/src/sensors.cpp.o
/usr/bin/c++  -DHAVE_SYSTEMD -DPID_FILE=\"/run/thinkfan.pid\" -DUSE_NVML -DUSE_YAML -DVERSION=\"1.3.1\" -I/home/morten/repos/thinkfan/include  -O2 -g -DNDEBUG   -Wall -std=c++1y -o CMakeFiles/thinkfan.dir/src/sensors.cpp.o -c /home/morten/repos/thinkfan/src/sensors.cpp
[ 55%] Building CXX object CMakeFiles/thinkfan.dir/src/message.cpp.o
/usr/bin/c++  -DHAVE_SYSTEMD -DPID_FILE=\"/run/thinkfan.pid\" -DUSE_NVML -DUSE_YAML -DVERSION=\"1.3.1\" -I/home/morten/repos/thinkfan/include  -O2 -g -DNDEBUG   -Wall -std=c++1y -o CMakeFiles/thinkfan.dir/src/message.cpp.o -c /home/morten/repos/thinkfan/src/message.cpp
[ 66%] Building CXX object CMakeFiles/thinkfan.dir/src/parser.cpp.o
/usr/bin/c++  -DHAVE_SYSTEMD -DPID_FILE=\"/run/thinkfan.pid\" -DUSE_NVML -DUSE_YAML -DVERSION=\"1.3.1\" -I/home/morten/repos/thinkfan/include  -O2 -g -DNDEBUG   -Wall -std=c++1y -o CMakeFiles/thinkfan.dir/src/parser.cpp.o -c /home/morten/repos/thinkfan/src/parser.cpp
[ 77%] Building CXX object CMakeFiles/thinkfan.dir/src/error.cpp.o
/usr/bin/c++  -DHAVE_SYSTEMD -DPID_FILE=\"/run/thinkfan.pid\" -DUSE_NVML -DUSE_YAML -DVERSION=\"1.3.1\" -I/home/morten/repos/thinkfan/include  -O2 -g -DNDEBUG   -Wall -std=c++1y -o CMakeFiles/thinkfan.dir/src/error.cpp.o -c /home/morten/repos/thinkfan/src/error.cpp
[ 88%] Building CXX object CMakeFiles/thinkfan.dir/src/yamlconfig.cpp.o
/usr/bin/c++  -DHAVE_SYSTEMD -DPID_FILE=\"/run/thinkfan.pid\" -DUSE_NVML -DUSE_YAML -DVERSION=\"1.3.1\" -I/home/morten/repos/thinkfan/include  -O2 -g -DNDEBUG   -Wall -std=c++1y -o CMakeFiles/thinkfan.dir/src/yamlconfig.cpp.o -c /home/morten/repos/thinkfan/src/yamlconfig.cpp
[100%] Linking CXX executable thinkfan
/usr/bin/cmake -E cmake_link_script CMakeFiles/thinkfan.dir/link.txt --verbose=1
/usr/bin/c++  -O2 -g -DNDEBUG  -rdynamic CMakeFiles/thinkfan.dir/src/thinkfan.cpp.o CMakeFiles/thinkfan.dir/src/config.cpp.o CMakeFiles/thinkfan.dir/src/fans.cpp.o CMakeFiles/thinkfan.dir/src/sensors.cpp.o CMakeFiles/thinkfan.dir/src/message.cpp.o CMakeFiles/thinkfan.dir/src/parser.cpp.o CMakeFiles/thinkfan.dir/src/error.cpp.o CMakeFiles/thinkfan.dir/src/yamlconfig.cpp.o  -o thinkfan  -lpthread -ldl -lyaml-cpp 
/usr/bin/ld: CMakeFiles/thinkfan.dir/src/yamlconfig.cpp.o: in function `YAML::Node::Scalar[abi:cxx11]() const':
/usr/local/include/yaml-cpp/node/impl.h:169: undefined reference to `YAML::detail::node_data::empty_scalar[abi:cxx11]()'
/usr/bin/ld: /usr/local/include/yaml-cpp/node/impl.h:169: undefined reference to `YAML::detail::node_data::empty_scalar[abi:cxx11]()'
/usr/bin/ld: /usr/local/include/yaml-cpp/node/impl.h:169: undefined reference to `YAML::detail::node_data::empty_scalar[abi:cxx11]()'
/usr/bin/ld: /usr/local/include/yaml-cpp/node/impl.h:169: undefined reference to `YAML::detail::node_data::empty_scalar[abi:cxx11]()'
/usr/bin/ld: /usr/local/include/yaml-cpp/node/impl.h:169: undefined reference to `YAML::detail::node_data::empty_scalar[abi:cxx11]()'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/thinkfan.dir/build.make:189: thinkfan] Error 1
make[2]: Leaving directory '/home/morten/repos/thinkfan/build'
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/thinkfan.dir/all] Error 2
make[1]: Leaving directory '/home/morten/repos/thinkfan/build'
make: *** [Makefile:130: all] Error 2
```

vmatare, 2022-08-25:
OK, that seems very weird, it's as if symbols are missing from your `libyaml-cpp`. If I check my `libyaml-cpp` for the symbols in question I can see that they exist:
```bash
# objdump -TC /usr/lib64/libyaml-cpp.so.0.7 | grep 'empty_scalar'
0000000000030cc0 g    DF .text  0000000000000091  Base        YAML::detail::node_data::empty_scalar[abi:cxx11]()
```
Could you try the same please?

mnissov, 2022-08-27:
Sorry for the late response, must have missed the notification.
```
# objdump -TC /usr/lib64/libyaml-cpp.so.0.7 | grep 'empty_scalar'
objdump: '/usr/lib64/libyaml-cpp.so.0.7': No such file
```

I'll try recompiling and see if it changes anything

mnissov, 2022-08-27:
Okay, so I removed the `yaml-cpp` I built from source, and installed through the package manager instead by `sudo apt install libyaml-cpp-dev`. Now it builds with `yaml-cpp` 0.6.2. I assume this is fine?

```
cmake -D CMAKE_BUILD_TYPE=Release ..
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Checking for module 'systemd'
--   Found systemd, version 245
-- Checking for module 'openrc'
--   No package 'openrc' found
-- Checking for module 'yaml-cpp'
--   Found yaml-cpp, version 0.6.2
-- Checking for module 'libatasmart'
--   No package 'libatasmart' found
-- Configuring done
-- Generating done
-- Build files have been written to: /home/morten/repos/thinkfan/build
❯ make
Scanning dependencies of target thinkfan
[ 11%] Building CXX object CMakeFiles/thinkfan.dir/src/thinkfan.cpp.o
[ 22%] Building CXX object CMakeFiles/thinkfan.dir/src/config.cpp.o
[ 33%] Building CXX object CMakeFiles/thinkfan.dir/src/fans.cpp.o
[ 44%] Building CXX object CMakeFiles/thinkfan.dir/src/sensors.cpp.o
[ 55%] Building CXX object CMakeFiles/thinkfan.dir/src/message.cpp.o
[ 66%] Building CXX object CMakeFiles/thinkfan.dir/src/parser.cpp.o
[ 77%] Building CXX object CMakeFiles/thinkfan.dir/src/error.cpp.o
[ 88%] Building CXX object CMakeFiles/thinkfan.dir/src/yamlconfig.cpp.o
[100%] Linking CXX executable thinkfan
[100%] Built target thinkfan
```

vmatare, 2022-08-27:
Yup, that's much better. As a general rule of thumb, stick with the distribution packages unless you have a very good reason not to use them.