vmatare/thinkfan #196: installed libyaml-cpp-dev not being found by cmake

dougalII, 2022-08-27:
Hey, 

Trying to install from source so i can get the sensors.chip functionality on my ubuntu 22.04 install.

I'v installed dependencies as described, Note: `pkgconfig` now seems to be called `pkg-config`
```bash
$ sudo apt install cmake g++ libyaml-cpp-dev pkg-config libsensors-dev libyaml-cpp-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
cmake is already the newest version (3.22.1-1ubuntu1).
cmake set to manually installed.
g++ is already the newest version (4:11.2.0-1ubuntu1).
libsensors-dev is already the newest version (1:3.6.0-7ubuntu1).
libyaml-cpp-dev is already the newest version (0.7.0+dfsg-8build1).
pkg-config is already the newest version (0.29.2-1ubuntu3).
0 to upgrade, 0 to newly install, 0 to remove and 0 not to upgrade.

```

but when running cmake i get:
```bash
$ cmake -D CMAKE_BUILD_TYPE=Release ..
-- Checking for module 'openrc'
--   No package 'openrc' found
-- Checking for module 'libatasmart'
--   No package 'libatasmart' found
CMake Error at CMakeLists.txt:72 (message):
  USE_YAML enabled but yaml-cpp not found.  Please install yaml-cpp[-devel]!

-- Configuring incomplete, errors occurred!
See also "/home/david/git/thinkfan/build/CMakeFiles/CMakeOutput.log".
```

I've tried both 0.6.0 and 0.7.0 of libyaml-cpp as #195 was also complaining about a similar isssue, but with no success.

one possible issue is libyaml-cpp seems to get installed in `/usr/lib/x86_64-linux-gnu` not `/usr/lib64/`
```
$ ls -ltrha /usr/lib/x86_64-linux-gnu/libyaml-cpp.so.0.7.0
-rw-r--r-- 1 root root 243K Mar 25 09:54 /usr/lib/x86_64-linux-gnu/libyaml-cpp.so.0.7.0
```

vmatare, 2022-09-18:
Hi @dougalII, sorry about the late reply... One reason is that I can't really help you much here. So you have `libyaml-cpp-dev` installed but cmake can't find it. At first glance, that makes no sense at all. So you could check whether `pkg-config` finds it:
```
pkg-config --exists yaml-cpp && echo 'yaml-cpp exists'
```
If that doesn't output "yaml-cpp exists", there is something wrong with your installation and you should check the contents of your distribution package for a `*.pc` file and see whether it's on pkg-config's search paths.

If you do get the output "yaml-cpp exists", then cmake has some problem and you'll need to debug it. Read its man page and pay particular attention to the `--debug-output` option and anything related to logging and tracing.

AikedeJongste, 2022-11-02:
Installing pkgconf fixed it for me.