vmatare/thinkfan #197: Features: hysteresis

mccarverone, 2022-09-17:
First thing, thank you very much for thinkfan, it works flawlessly in my setup.

However I would like to suggest a feature. I do a lot of compiling jobs and temperatures might change rather erratically in short periods of time. The fans are thus often speeding up and down. This might be annoying in the long run.

A much smoother behavior would be to slow down the fans only after a specific custom period of time (e.g. 5 seconds) has passed during which temperatures has been consistently below the current fan speed level.

This is only a cosmetic change, on the other hand it would be totally safe since it would imply a higher fan speed.

Ever considered something like this?

Cheers

vmatare, 2022-09-18:
Thanks for the kind words @mccarverone. Of course I've thought about hysteresis, and my conclusion has always been that it should be expressed in terms of temperature and not time. That's why thinkfan enforces a **minimum** hysteresis of 1°C between fan speeds.

Another tool you have available if you have small thermal mass and your temperatures are changing too rapidly is the biasing functionality with the `-b<BIAS>` commandline option. Set a positive bias to exaggerate rapid temperature changes, and a negative bias to smooth them out.

So first of all I'd like to know how you're using these two, i.e. what is your `levels:` config and how (if at all) are you using the `-b` option?

mccarverone, 2022-09-18:
These are my current levels:
```
levels:
  - [0, 0, 55]
  - [1, 50, 65]
  - [2, 58, 70]
  - [3, 63, 75]
  - [4, 68, 82]
  - [5, 75, 85]
  - [6, 78, 88]
  - [7, 83, 32767]
```
I'm already trying to spread the intervals by lowering the lower bounds.
As far as bias is concerned, it should not act as hysteresis because it would be the same as if I changed the current bounds, am I right?
As a matter of fact I don't even know if time hysteresis could be useful, but I would like to try it.
In my current setup maybe I have too much levels set up and this might produce more fan changes.

mccarverone, 2022-09-20:
I am studying thinkfan a bit more and I now understand that "bias" is different from "correction".

Bias amplifies changes in temperatures so that fan levels are escalated more rapidly. It can be set via commandline but I cannot find a way to set a bias via the config file.

Also, the default value is 5 in `thinkfan -h` while it is 15 in `man thinkfan`. This should be reviewed.

vmatare, 2022-09-21:
> Bias amplifies changes in temperatures so that fan levels are escalated more rapidly.
Yes, but you can also have a negative bias to smooth out the temperature curve.

> It can be set via commandline but I cannot find a way to set a bias via the config file.
Correct, it's a command-line option only.

> This is necessary in order to use thinkfan as a service.
Incorrect, every service manager should provide some way for users to set the command line options of a given service. If you're using systemd, you can simply run `systemctl edit thinkfan`.

> Also, the default value is 5 in `thinkfan -h` while it is 15 in `man thinkfan`. This should be reviewed.
You must be using a very old version of thinkfan. In current versions the default is 0, and that is correctly represented in the man page. The help screen however still incorrectly states that the default is 5, so thanks for the heads-up anyways ;-)

mccarverone, 2022-09-22:
At the moment I am on Gentoo which uses OpenRC as init system, and I can't find a way to properly alter commandline arguments (apart from modifying the start script itself, which I wouldn't like to do).

I was on thinkfan 1.2.2 but I now upgraded to 1.3.1. As a side note I still find the same incongruent bias default levels (5 in help and 15 in man page).

I think I found a (obviously) more stable configuration with fewer and wider levels:
```
levels:
  - [0, 0, 55]
  - [1, 50, 65]
  - [3, 60, 75]
  - [5, 70, 85]
  - [7, 80, 32767]
```
However I will still experiment with bias option as a personal interest. In the default state I find that a bias is applied:
```
Temperatures(bias): 41(0), 41(0), 38(0), 57(9), 58(10), 50(0), 54(0) -> Fans: level 3
Temperatures(bias): 41(0), 41(0), 38(0), 67(15), 67(13), 53(4), 51(0) -> Fans: level 5
Temperatures(bias): 41(0), 41(0), 37(0), 56(0), 55(0), 51(0), 55(6) -> Fans: level 3
```
This means that default configuration is not 0 bias in this case?

bhundven, 2022-11-17:
seems that you just update the command args in the init file: https://wiki.gentoo.org/wiki/Fan_speed_control/thinkfan#Start_thinkfan_as_a_service