vmatare/thinkfan #199: ERROR: Unhandled std::bad_optional_access: bad optional access.

ciminoV, 2022-09-30:
Hi,
Installed following the readme instructions with default parameters. 
Thinkfan.service throws the following error on start:
```
ERROR: Unhandled std::bad_optional_access: bad optional access.
errno = No such file or directory.
Backtrace:
/usr/local/sbin/thinkfan(+0x49be4) [0x556e75581be4]
/usr/local/sbin/thinkfan(+0x26cf8) [0x556e7555ecf8]
/usr/lib/libstdc++.so.6(+0xa5cfc) [0x7fc727e9dcfc]
/usr/lib/libstdc++.so.6(+0xa5d69) [0x7fc727e9dd69]
/usr/lib/libstdc++.so.6(+0xa5fcd) [0x7fc727e9dfcd]
/usr/local/sbin/thinkfan(+0x2515d) [0x556e7555d15d]
/usr/local/sbin/thinkfan(+0x24135) [0x556e7555c135]
/usr/local/sbin/thinkfan(thinkfan::Driver::try_init()+0x3f) [0x556e75573e5f]
/usr/local/sbin/thinkfan(thinkfan::Config::try_init_driver(thinkfan::Driver&) const+0x2f) [0x556e75565a8f]
/usr/local/sbin/thinkfan(thinkfan::Config::init_sensors(thinkfan::TemperatureState&) const+0x4f) [0x556e75565b2f]
/usr/local/sbin/thinkfan(main+0x244) [0x556e75561c24]
/usr/lib/libc.so.6(+0x23290) [0x7fc727b2c290]
/usr/lib/libc.so.6(__libc_start_main+0x8a) [0x7fc727b2c34a]
/usr/local/sbin/thinkfan(_start+0x25) [0x556e755628b5]
```

Although it works fine on the command line with:
```
thinkfan -n
```

This is happening on a Thinkpad T480 on Archlinux.

thinkpad.yaml
```
sensors:
  - tpacpi: /proc/acpi/ibm/thermal
    indices: [0]
    max_errors: 10

fans:
  - tpacpi: /proc/acpi/ibm/fan
    max_errors: 10

levels:
- [0, 0, 50]
- [1, 45, 65]
- ["level auto", 60, 85]
- ["level disengaged", 80, 255]
```

ciminoV, 2022-10-06:
Up

vmatare, 2022-11-20:
This is probably related to #206...

vmatare, 2022-11-20:
I've fixed an issue with `tpacpi` sensor initialization in the latest master, could you please build that and report whether it fixes your problem?

ciminoV, 2022-11-20:
Problem fixed! Thank you

closed 2022-11-20 by ciminoV