vmatare/thinkfan #222: temperature input value of feature 'temp2' of chip 'thinkpad-isa-0000' is unavailable

plkrnr, 2023-01-15:
Hi,
small issue I think, but it's a confusing one. Using latest code in the repo.

(relevant) `sensors`:
```
thinkpad-isa-0000
Adapter: ISA adapter
fan1:           0 RPM
CPU:          +40.0°C  
GPU:              N/A  
temp3:        +40.0°C  
temp4:         +0.0°C  
temp5:        +40.0°C  
temp6:        +40.0°C  
temp7:        +40.0°C  
temp8:            N/A 
```

(relevant) config:
```
sensors:
  - chip: thinkpad-isa-0000
    ids: [ CPU, GPU, temp3, temp4, temp5, temp6, temp7, temp8 ]
```

output:
```
sudo thinkfan -nv
ERROR: temperature input value of feature 'temp2' of chip 'thinkpad-isa-0000' is unavailable: Can't read
```

expected output: ERROR: temperature input value of feature 'GPU'
if I would but _GPU_ at the end of the ids array it would also say 'temp2'
(removing GPU and temp8 from my config fixed the issue for me)

vmatare, 2024-10-15:
That sounds very weird. From a code review (currently have no hardware available to reproduce this) I'd say this must be a libsensors bug, because we first lookup the feature by name (i.e. `GPU` in this case), but then continue on with the libsensors representation of the feature. In the error message, we then use whatever libsensors says the offending feature's name is. So it looks to me like libsensors just has the wrong feature name in its internal data structure.

Detzi88, 2025-03-29:
I had the exact issue on a P14s G5 AMD. My sensors output:
```bash
thinkpad-isa-0000
Adapter: ISA adapter
fan1:           0 RPM
fan2:           0 RPM
CPU:          +49.0°C  
GPU:              N/A  
temp3:        +49.0°C  
temp4:         +0.0°C  
temp5:            N/A  
temp6:        +49.0°C  
temp7:        +49.0°C  
temp8:            N/A
```

And for me as well with:
```yaml
sensors:
  - chip: thinkpad-isa-0000
    ids: [ CPU, GPU, temp3, temp4, temp5, temp6, temp7, temp8]
```
```bash
sudo thinkfan -n
ERROR: temperature input value of feature 'temp2' of chip 'thinkpad-isa-0000' is unavailable: Can't read
```
which is just printing the wrong feature 'temp2' but otherwise is correct. Removing the N/A sensors
```yaml
sensors:
  - chip: thinkpad-isa-0000
    ids: [ CPU, temp3, temp4, temp6, temp7]
```
 solves the issue.

4l1fe, 2025-06-01:
And me had the issue, removing `GPU, temp8` in the `sensors.ids` attribute solved it.