vmatare/thinkfan #229: Invalid sensor entry

nozwock, 2023-06-15:
```
ERROR: /etc/thinkfan.yaml:56:
  - chip: thinkpad-isa-0000
    ^
Invalid sensor entry.
```

Using the [sample yaml config](https://github.com/vmatare/thinkfan/blob/8a844b96cf7b3aaa87a85d317615851faba69284/examples/thinkfan.yaml).

vmatare, 2023-06-16:
First thing I'd normally suspect is that `USE_LM_SENSORS` has been disabled at compile time. But I don't know how to find out if that's the case or not.

vmatare, 2023-06-16:
I guess I *could* implement checks for disabled features that would give a more helpful error message in that kind of situation...

nozwock, 2023-06-16:
> First thing I'd normally suspect is that `USE_LM_SENSORS` has been disabled at compile time. But I don't know how to find out if that's the case or not.

I'll try compiling myself with that flag.

nozwock, 2023-06-16:
This worked, it was missing the `USE_LM_SENSORS` flag.
` cmake .. -DCMAKE_BUILD_TYPE:STRING=Release -DUSE_LM_SENSORS:BOOL=ON`

<details><summary>AUR PKGBUILD Fix</summary>
<p>

```
# Maintainer: Barry Smith <brrtsm at gmail.com>
# Contributor: nblock <nblock [/at\] archlinux DOT us>
# Contributor: Frederik AlkÃ¦rsig (FALKER) <havnelisten AT gmail.com>

pkgname=thinkfan
pkgver=1.3.1
pkgrel=1
pkgdesc="A minimalist fan control program. Supports the sysfs hwmon interface and thinkpad_acpi"
url="https://github.com/vmatare/thinkfan"
arch=('i686' 'x86_64')
makedepends=('cmake')
depends=('libatasmart' 'yaml-cpp')
optdepends=('lm_sensors: hwmon support')
backup=('etc/systemd/system/thinkfan.service.d/override.conf')
license=('GPL')
source=("$pkgname-$pkgver.tar.gz::https://github.com/vmatare/$pkgname/archive/$pkgver.tar.gz"
        'thinkfan.service'
        'thinkpad_acpi.conf')
sha256sums=('9466c8c82b7c4333b280fa66445ab26185ffbb4aada6bcb4a164eed742f8d78c'
            '625730e33fae251f3a1cc8259b222089fb45ada08e9cf5c816ea1cde6560e989'
            '28874576641b3cc1aa486e1bc9ed40eadfe8b13d6d3014016e31fedc5dc5195f')
install=thinkfan.install

build() {
  cd "$srcdir/$pkgname-$pkgver"

  cmake . -DCMAKE_INSTALL_PREFIX=/usr -DUSE_ATASMART:BOOL=ON -DUSE_LM_SENSORS:BOOL=ON
  make
}

package() {
  cd "$srcdir/$pkgname-$pkgver"

  install -d -m755 "$pkgdir"/usr/{bin,lib/{modprobe.d,systemd/system},share/{doc/"$pkgname"/{,examples},man/man{1,5}}}
  install -d -m755 "$pkgdir"/etc/systemd/system/thinkfan.service.d
  install -D -m755 "$pkgname" "$pkgdir"/usr/bin
  install -D -m644 examples/* "$pkgdir"/usr/share/doc/"$pkgname"/examples
  install -D -m644 "$pkgname".1 "$pkgdir"/usr/share/man/man1
  install -D -m644 "$pkgname".conf.5 "$pkgdir"/usr/share/man/man5
  install -D -m644 "$srcdir/$pkgname".service rcscripts/systemd/thinkfan-{sleep,wakeup}.service "$pkgdir"/usr/lib/systemd/system
  install -D -m644 rcscripts/systemd/override.conf "$pkgdir"/etc/systemd/system/thinkfan.service.d
  install -D -m644 "$srcdir"/thinkpad_acpi.conf "$pkgdir"/usr/lib/modprobe.d
}

# vim:set ts=2 sw=2 noet:
```

</p>
</details>

nozwock, 2023-06-16:
Now I'm getting
```
ERROR: /etc/thinkfan.yaml:136:
  - hwmon: /sys/class/hwmon/hwmon0/pwm1
  ^
When multiple fans are configured, each must have its own 'levels:' section.
```
What should I do?

nozwock, 2023-06-16:
> But I don't know how to find out if that's the case or not if you're using an arch package.

You could just view the PKGBUILD here https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=thinkfan
But I just realized that the package was not submitted by you, so ye.

nozwock, 2023-06-16:
I commented out some more options which were leading to issues but it still doesn't work, I'm getting how to get thinkfan to work.

<details><summary>thinkfan.yaml</summary>
<p>

```
##############################################################################
# thinkfan Example Config File
# ============================
#
# Please read the config manpage thinkfan.conf(5) before playing around with
# this.
#
# This is NOT a working config file that can just be copied. It is only meant
# to give a rough idea what can be done. In particular, don't copy & paste the
# fan speed config! Think about what you're doing.
#
# If you don't know what temperatures are right for your system, you should
# not be using thinkfan!
##############################################################################

##############################################################################
# Sensor Drivers and Temperature Inputs
# =====================================
#
# ATTENTION: The order in which sensors are specified here is significant when
# specifying the fan speeds further below!
#
# There are multiple ways in which a temperature input can be specified. An
# example for each is given below.
#
# The "correction:" and "optional:" keywords may be specified on any type of
# sensor.

sensors:
  # LM Sensors
  # ==========
  # Temperatures can be read directly from Linux drivers through the LM sensors.
  #
  # To configure this, install "lm-sensors" and "libsensors", then
  # run "sensors-detect", then run "sensors".
  # To build thinkfan from sources, you'll also need to install "libsensors-dev"
  # or equivalent package for your distribution.
  #
  # For example, the following output of "sensors":
  # ...
  # thinkpad-isa-0000
  # Adapter: ISA adapter
  # fan1:        2618 RPM
  # fan2:        2553 RPM
  # CPU:          +63.0 C  
  # GPU 1:        +55.0 C  
  # temp3:        +68.0 C  
  # temp4:         +0.0 C  
  # temp5:        +60.0 C  
  # temp6:        +64.0 C  
  # temp7:        +67.0 C  
  # temp8:         +0.0 C  
  # ...
  # would result in the following configuration:
  - chip: zenpower-pci-00c3
    ids: [ Tctl ]

  # hwmon: Full path to a temperature file (single sensor).
  # =======================================================
  # Disadvantage is that the index in "hwmon0" depends on the load order of
  # the driver modules, which may change across bootups on some systems.
  # - hwmon: /sys/class/hwmon/hwmon0/temp1_input

  # hwmon: Path to a complete driver folder
[... 120 lines cut ...]
# The speed values used here range from 0 to 255, which is valid for the PWM
# control files used by hwmon-based drivers.
#
# The temperatures specified in upper_limit and lower_limit apply to the
# sensors in the same order in which they were specified in the "sensors:"
# section above, and their length must match the total number of sensors that
# have been configured.
#
levels:
  - speed: 0
    upper_limit: [50, 50, 50]

  - speed: 100
    lower_limit: [45, 45, 45]
    upper_limit: [65, 65, 65]

  - speed: 255
    lower_limit: [60, 60, 60]

##############################################################################

```

</p>
</details>

nozwock, 2023-06-16:
Ok, now it's working.
```sh
$ sudo thinkfan
[sudo] password for nozwock:

Daemon PID: 194920
```

<details><summary>thinkfan.yaml</summary>
<p>

```
##############################################################################
# thinkfan Example Config File
# ============================
#
# Please read the config manpage thinkfan.conf(5) before playing around with
# this.
#
# This is NOT a working config file that can just be copied. It is only meant
# to give a rough idea what can be done. In particular, don't copy & paste the
# fan speed config! Think about what you're doing.
#
# If you don't know what temperatures are right for your system, you should
# not be using thinkfan!
##############################################################################

##############################################################################
# Sensor Drivers and Temperature Inputs
# =====================================
#
# ATTENTION: The order in which sensors are specified here is significant when
# specifying the fan speeds further below!
#
# There are multiple ways in which a temperature input can be specified. An
# example for each is given below.
#
# The "correction:" and "optional:" keywords may be specified on any type of
# sensor.

sensors:
  # LM Sensors
  # ==========
  # Temperatures can be read directly from Linux drivers through the LM sensors.
  #
  # To configure this, install "lm-sensors" and "libsensors", then
  # run "sensors-detect", then run "sensors".
  # To build thinkfan from sources, you'll also need to install "libsensors-dev"
  # or equivalent package for your distribution.
  #
  # For example, the following output of "sensors":
  # ...
  # thinkpad-isa-0000
  # Adapter: ISA adapter
  # fan1:        2618 RPM
  # fan2:        2553 RPM
  # CPU:          +63.0 C  
  # GPU 1:        +55.0 C  
  # temp3:        +68.0 C  
  # temp4:         +0.0 C  
  # temp5:        +60.0 C  
  # temp6:        +64.0 C  
  # temp7:        +67.0 C  
  # temp8:         +0.0 C  
  # ...
  # would result in the following configuration:
  - chip: zenpower-pci-00c3
    ids: [ Tctl ]

  # hwmon: Full path to a temperature file (single sensor).
[... 126 lines cut ...]
# The speed values used here range from 0 to 255, which is valid for the PWM
# control files used by hwmon-based drivers.
#
# The temperatures specified in upper_limit and lower_limit apply to the
# sensors in the same order in which they were specified in the "sensors:"
# section above, and their length must match the total number of sensors that
# have been configured.
#
levels:
  - speed: 0
    upper_limit: [50, 50, 50]

  - speed: 100
    lower_limit: [45, 45, 45]
    upper_limit: [65, 65, 65]

  - speed: 255
    lower_limit: [60, 60, 60]

##############################################################################

```

</p>
</details>

nozwock, 2023-06-16:
Ok, nvm I manged to make it work.

```yaml
levels:
  # - [0, 0, 50]
  - ["level 2", 0, 50]
  - ["level 4", 50, 70]
  - ["level disengaged", 70, 255]
```

Thanks for the project, you can close the issue now if you'd like.

jpsutton, 2023-09-17:
It appears that USE_LM_SENSORS wasn't added until after the current release (1.3.1) was tagged. That being said, the fix applied to the AUR PKGBUILD mentioned above doesn't work; you'll see the following during the build process...

```
CMake Warning:
  Manually-specified variables were not used by the project:

    USE_LM_SENSORS
```

@vmatare it's been a few years since a release. Is there anything holding that up?

nozwock, 2023-09-17:
@jpsutton The flag was probably added after the latest tagged release. It's available on `main`.
https://github.com/vmatare/thinkfan/blob/b1ad819e4ec79776cb5ccc61e2206a0c5d506ef5/CMakeLists.txt#L47-L50

jpsutton, 2023-09-17:
Yes, that's what I was saying. The question is, is there anything holding up a new release version tag? Having a newer tagged version would cause this feature to make it's way into distro packages (or in my case, at least into AUR), so that we don't have to manually build from `main`.

donalffons, 2024-03-23:
Thank you for this project! @vmatare it would be great for all arch users if a new tag / minor version could be released including that change.

donalffons, 2024-03-23:
Just found out that [thinkfan-git](https://aur.archlinux.org/packages/thinkfan-git) exists in AUR, which does solves this problem for me. Still, I think it would be great, if this fix would be released.

techeddie, 2024-04-06:
> Just found out that [thinkfan-git](https://aur.archlinux.org/packages/thinkfan-git) exists in AUR, which does solve this problem for me. Still, I think it would be great, if this fix would be released.

With thinkfan-git it is working for me as well with the thinkfan.yaml file.

bakkerthehacker, 2024-08-20:
Still waiting for a real release for debian/ubuntu.  The [sources](https://salsa.debian.org/debian/thinkfan) here are still missing lmsensors support.

I did get a .deb built using the following steps:

- `apt-get source thinkfan`
- manually replace the archive with master
- remove patches 004 through 007
- `dpkg-source --no-check -x thinkfan_1.3.1-4build1.dsc`
- `dch -i` with a modified version number
- `debuild -us -uc -b`

Not great, and its using custom release numbers but it does work with lmsensors now.

tova-cup, 2024-08-29:
Hi @bakkerthehacker 

Would you mind sharing your compiled .deb ?

I'm struggling to reproduce your process.