vmatare/thinkfan #230: [ThinkPad E540] Thinkfan change the levels, but not the actual speed

Drommer, 2023-06-26:
I think the problem is more with the driver than with thinkfan itself. However, maybe someone has encountered the same problem and can help.

I have set up getting the temperature from the sensors and changing the level in `/proc/acpi/ibm/fan`, but it has no effect on the actual speed, at any level I have `speed: 0`.

`/etc/thinkfan.conf`:
> sensors:
>   - hwmon: /sys/class/hwmon
>     name: coretemp
>     indices: [1]
> 
>   - hwmon: /sys/class/hwmon
>     name: acpitz
>     indices: [1]
> 
>   - nvml: 01:00.0
>     optional: true
> 
> fans:
>   - tpacpi: /proc/acpi/ibm/fan
> 
> levels:
>   - speed: 0
>     upper_limit: [50, 50, 50]
> 
>   - speed: 3
>     lower_limit: [45, 45, 45]
>     upper_limit: [65, 65, 65]
> 
>   - speed: 7
>     lower_limit: [55, 55, 55]
>     upper_limit: [75, 75, 75]
> 
>   - speed: "level full-speed"
>     lower_limit: [65, 65, 65]
>     upper_limit: [90, 90, 90]

`systemctl status thinkfan`:
> thinkfan[465]: Temperatures(bias): 66(0), 67(0), 0(0) -> Fans: level 7
> thinkfan[465]: Temperatures(bias): 52(0), 54(0), 0(0) -> Fans: level 3

`/proc/acpi/ibm/fan`
> status: enabled
> speed: 0
> level: 3
> commands: level <level> (<level> is 0-7, auto, disengaged, full-speed)

I also create the file `/etc/modprobe.d/thinkpad_acpi.conf` with line
> options thinkpad_acpi fan_control=1 experimental=1

What am I doing wrong? Is this laptop model so old that it does not support tpacpi control?

Lillecarl, 2023-09-07:
as root try running this command:
```
echo "level 7" > /proc/acpi/ibm/fan
```
If your fan doesn't start going like crazy it's the kernel, if it does it's thinkfan.

dfischer02, 2023-09-08:
What needs to be done if it is the kernel?

Lillecarl, 2023-09-11:
@dfischer02 Make sure you're on the latest EC firmware. If that doesn't work then a bug must be filed against the kernel, and Lenovo people must bring attention to it.

You're essentially on your own 😢

alemoh1234, 2023-10-05:
I have come across a similar problem but with an X380. I searched a lot for a way to control this fan in windows, but apparently there was none (only an abandoned attempt to reverse engineer the EC registers).

I then found this project for linux, but my fan does not seem to support manual control, can you corroborate that for me?
`/proc/acpi/ibm/fan` output:
```
status:         enabled
speed:          0
level:          auto
```