vmatare/thinkfan #233: yaml-cpp 0.8.0 compatibility

k1gen, 2023-09-17:
yaml-cpp updated to 0.8.0 on arch, so thinkfan stopped working:
`/usr/bin/thinkfan: error while loading shared libraries: libyaml-cpp.so.0.7: cannot open shared object file: No such file or directory`

nozwock, 2023-09-17:
Have you tried the `main` build of thinkfan? Does it also have the incompatibility?

nozwock, 2023-09-17:
You may also try symlinking the `0.8.0` `libyaml` to `libyaml-cpp.so.0.7`, would probably work.

k1gen, 2023-09-17:
haven't tried the `main` build yet, am using v1.3.1 from arch repos. also, while the lib symlink is an ugly decision, it works. `sudo ln -s libyaml-cpp.so.0.8 /usr/lib/libyaml-cpp.so.0.7` fixed the issue for me. waiting for the upstream fix anyway

nozwock, 2023-09-17:
Ok, so all you have to do is rebuild the binary with the updated dependencies.
```
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.1") 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Checking for module 'systemd'
--   Found systemd, version 254
-- Checking for module 'openrc'
--   Package 'openrc', required by 'virtual:world', not found
-- Checking for module 'yaml-cpp'
--   Found yaml-cpp, version 0.8.0
-- Checking for module 'libatasmart'
--   Found libatasmart, version 0.19
-- Configuring done (0.6s)
-- Generating done (0.0s)
-- Build files have been written to: /tmp/thinkfan
```
So you either need to do the lib symlinking or have to rebuild the binary.

bssb, 2023-09-17:
This is pretty much a duplicate of #74, #162, #169.

It's just the way AUR packaging works. If your AUR helper program doesn't automatically rebuild a package when one of its dependency is updated, you have to rebuild the package manually. I don't know of any AUR helpers that can do that btw, I've always rebuilt manually.

JohnnyNich, 2023-10-05:
In this case I've found that even with a yay cleanbuild (recommended solution in the issue #169 ) I receive the same error. Is it required to uninstall an AUR version of Thinkfan and then build it manually from source?

closed 2024-10-15 by vmatare