vmatare/thinkfan #234: setting RPM that corresponds to the speed levels manualy

smautner, 2023-09-26:
Hi, Haven't found an issue on this obvious item so here i go. 

I have an T14 gen 4 AMD, level 1 maps to a fan speed that is too loud, so was wondering if there is a solution to this.
Is there an easy way to map the speed levels to actual fan-speeds?

smautner, 2023-09-27:
Thank you. Working without rpm mapping, I haven't found a setting i like. I also figured that increasing the temperature range for level 0 is important, but going auto never occurred to me.

vmatare, 2024-10-15:
On thinkpads the embedded controller only exposes certain fixed levels to userspace. If you use the hwmon interface, values theoretically range from 0 to 255, but in my experience, thinkpad ECs also quantize these into a small number of fixed levels. So yeah, the best advice to get some smoothness is to use `level auto` and only configure explicit levels at the top and bottom end of the temperature range.

closed 2024-10-15 by vmatare