vmatare/thinkfan #239: Thinkfan does not resume after wake from suspend

DikshitRJ, 2024-03-19:
When i wake my laptop from suspend, thinkfan shows the error "ERROR: Lost sensor read_temps: Failed to read temperature(s) from /sys/devices/virtual/thermal/thermal_zone6/hwmon7/temp1_input: No data available". thank you in advance

rustyx, 2024-04-07:
That could be the wifi adapter. Try `cat /sys/devices/virtual/thermal/thermal_zone6/hwmon7/name`.
The iwlwifi adapter takes a while to wake up and its thermal data is unavailable for a few seconds after a resume.

vitaly-zdanevich, 2024-06-20:
After wakeup it crashes on my system (Gentoo with OpenRC), in logs I see:

> ERROR: Lost sensor read_temps_: Failed to read temperature(s) from /sys/class/hwmon/hwmon1/temp1_input: No such device or address