vmatare/thinkfan #244: hwmon model is an invalid keyword

bhundven, 2024-06-13:
In https://github.com/vmatare/thinkfan/pull/211, I implemented the hwmon `model` config option.
In the latest git build, `model` is no longer valid:

config:
```yaml
  - hwmon: /sys/class/hwmon
    name: nvme
    model: WD_BLACK SN770 1TB
    indices: [1]
```

log:
```
Jun 13 13:12:57 mill systemd[1]: Starting thinkfan.service - thinkfan 2.0.0...
Jun 13 13:12:57 mill thinkfan[16494]: ERROR: /etc/thinkfan.conf:10:
                                          model: WD_BLACK SN770 1TB
                                          ^
                                      Invalid keyword.
```

This is on the current head: b1ad819e4ec79776cb5ccc61e2206a0c5d506ef5

vmatare, 2024-10-15:
Should be fixed in latest master.

bhundven, 2024-10-15:
Tested on Fedora Workstation 41.

bhundven, 2024-10-15:
So this is "kind of fixed". Let me get some more info an elaborate.

bhundven, 2024-10-15:
Here is my config:
```yaml
---
sensors:
  - tpacpi: /proc/acpi/ibm/thermal
  - nvml: 01:00.0
  - hwmon: /sys/class/hwmon
    name: coretemp
    indices: [1, 2, 3, 4, 5, 6, 7, 8, 9]
  - hwmon: /sys/class/hwmon
    name: nvme
    indices: [1]
    model: WD_BLACK SN770 1TB

fans:
  - tpacpi: /proc/acpi/ibm/fan

levels:
  - ["level 0", 0, 28]
  - ["level auto", 25, 55]
  - ["level 6", 50, 75]
  - ["level 7", 70, 110]
  - ["level full-speed", 105, 9000]
```

I have an internal nvme named:
```sh
$ cat /sys/class/hwmon/hwmon3/name 
nvme
$ cat /sys/class/hwmon/hwmon3/device/model 
WD_BLACK SN770 1TB
```

And sometimes an external nvme for backup (thunderbolt 4):
```sh
$ cat /sys/class/hwmon/hwmon4/name 
nvme
$ cat /sys/class/hwmon/hwmon4/device/model 
WDS200T1XHE-00AFY0
```

With both plugged in with the above config, I get:
```
ERROR: hwmon sensor driver: /sys/class/hwmon: Found multiple hwmons with this name:  /sys/class/hwmon/hwmon4 /sys/class/hwmon/hwmon3
```

If I comment out `name`:
```yaml
  - hwmon: /sys/class/hwmon
    # name: nvme
    indices: [1]
    model: WD_BLACK SN770 1TB
```
Then this works. `indices` seems to be needed to be set.